xArm-Developer/xarm_ros #98: XArm 7 Spoofing End Effector RS485 Communication with Arduino

MeticulousFishFrier, 2022-02-08:
I'm trying to control a custom 3 pin TTL gripper with the XArm 7 via ROS. I'm using an Arduino Nano and an RS485 to TTL converter to handle the Modbus RTU protocol request and response such that I can control the 3 pin TTL gripper. This way I don't have to changes the underlying driver code of xarm_ros and would be able to control the custom gripper just as if I had the official XArm gripper. After physically connecting the Arduino to the RS485 to TTL converter and the converter to the RS485 port, I am unable to decipher completely the messages that the arm sends to the Arduino (which I'm reading using Arduino's serial.read() function). 
For example, when performing `rosservice call /xarm/set_tool_modbus [0x01,0x06,0x00,0x0A,0x00,0x03] 6`. I received in binary this,`1 10100110 11111111 1000 10000011 1 10100110 11111111 1000 10000011 11111111 `, in my Arduino Serial Monitor.  When performing `rosservice call /xarm/set_tool_modbus [0x08,0x06,0x01,0x00,0x00,0x01] 6`. I received in binary this,`1000 110 1 11100111 11111111 11111111 11111011 11111011 1000 110 1 10011101 10011111 11111111  `. For both examples, there seems to be additional redundant data that mirrors what came before. For the second example, the first three numbers of output (1000 110 1 ) seem to match up with the first three of the input (0x08,0x06,0x01), but the rest of the output seems to be jumbled. Any information that can help me make sense of what `set_tool_modbus` does exactly and how I can proceed to make the Arduino act like an official XArm gripper would be greatly appreciated. Attached is a picture of my setup.
[image]

vimior, 2022-02-09:
Hi, MeticulousFishFrier, What baud rate is the Arduino receiving?

penglongxiang, 2022-02-10:
Hi, please try to use **HEX format** to print the data stream received. Moreover, check if the tool modbus baudrate was correctly set by
```
rosservice call /xarm/get_tgpio_modbus_baudrate
```

MeticulousFishFrier, 2022-02-15:
Hi, 

I tested the Modbus's baudrate with `rosservice call /xarm/get_tgpio_modbus_baudrate` and it is set to the same baud rate as the Arduino (115200).

 These are the results from the command `rosservice call /xarm/set_tool_modbus [0x01,0x06,0x00,0x0A,0x00,0x03] 6`

`1 A6 FF 7D FF FF 1 A6 FF 7D FF `

although sending the same command right after gives 

`1 E6 FF 7D FF 1 E6 FF 7F FF`

These are the result from the command `rosservice call /xarm/set_tool_modbus [0x08,0x06,0x01,0x00,0x00,0x01] 6`

`8 E6 FB FF FF 8 E6 ED FF FF FF`

and if the command was sent again,

`8 E6 ED FF FF FF 8 E6 FB FF FF`

For reference, this is my Arduino code 
```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);                           // set serial communication baudrate 
}

void loop() {
  // put your main code here, to run repeatedly:

if(Serial.available())                     // If serial data is available then enter into while loop
  {
    Serial.print(Serial.read(),HEX);                     // Print Data
    Serial.print(" ");
  }   
}
```

I wonder if the problem stems from the way I'm reading the incoming signals with the Arduino because I'm clearly getting distinct results from two different commands, but the results just aren't the same as what I presume would be sent.

penglongxiang, 2022-02-16:
Sorry we do not have Arduino hardware and development experience. You may verify the outcome with serial port debugging tools at PC first, I have tested with cutecom, configure the serial port as follows, the received bytes were correct.

[image]