xArm-Developer/xarm_ros #99: error about using "d435i_xarm_auto_calib.launch"

youdaorobot, 2022-02-13:
Hello，I have a question and need help.
I installed d435i to the end of Xarm6，and follow the step,launch "d435i_xarm_auto_calib.launch", there is no error,and aruco result can show the coordinate about calibration plate.Then I use the GUI,"Check starting pose ->next pose->Plan->Execute",everthing is ok.
Until I click "Take sample",there is error as follow: 
[ERROR] [1644746643.988081]: Error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist. 
['Traceback (most recent call last):\n', '  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 632, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/youdao/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 88, in take_sample\n    self.sampler.take_sample()\n', '  File "/home/youdao/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 88, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/youdao/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 78, in _get_transforms\n    Duration(10))\n', '  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 87, in lookup_transform\n    return self.lookup_transform_core(target_frame, source_frame, time)\n', 'LookupException: "camera_marker" passed to lookupTransform argument source_frame does not exist. \n']
Traceback (most recent call last):
  File "/home/youdao/ros_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 132, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/youdao/ros_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 76, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 522, in call
    responses = transport.receive_once()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 735, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 360, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 343, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/xarm_realsense_handeyecalibration_eye_on_hand/take_sample] responded with an error: error processing request: "camera_marker" passed to lookupTransform argument source_frame does not exist.
[xarm_realsense_handeyecalibration_eye_on_hand/namespace_youdao_25390_6294775491585827593_rqt-14] process has died [pid 25566, exit code -6, cmd /home/youdao/ros_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_easy_handeye __name:=namespace_youdao_25390_6294775491585827593_rqt __log:=/home/youdao/.ros/log/33de4e52-8cb4-11ec-9694-3c970e6b3238/xarm_realsense_handeyecalibration_eye_on_hand-namespace_youdao_25390_6294775491585827593_rqt-14.log].
log file: /home/youdao/.ros/log/33de4e52-8cb4-11ec-9694-3c970e6b3238/xarm_realsense_handeyecalibration_eye_on_hand-namespace_youdao_25390_6294775491585827593_rqt-14*.log

penglongxiang, 2022-02-14:
Hi, please confirm from the camera image that the printed marker can be fully seen and detected by the camera, if successfully detected, an x-y-z coordinate system will appear at the center of marker, as well as detected marker ID, like this:
[image]

(source: https://github.com/pal-robotics/aruco_ros)

If detection fails, the "`camera_marker`"  frame may not be generated.

goyalvarun02, 2022-08-12:
Hi @youdaorobot  ,

I have also faced this issue while taking sample , and you just need to be careful and check whether Aruco Marker is completely visible or not . If not it will generate an error if visible then able to take sample successfully.

Thanks