xArm-Developer/xarm_ros #100: How to use move_servo with robot visualization in rviz?

petpetpeter, 2022-03-18:
According to
> https://github.com/xArm-Developer/xarm_ros/blob/master/examples/ReadMe.md
```
# 2. Servo_Cartesian (streamed Cartesian trajectory)
**PLEASE NOTE:** If you are already using **Moveit** to control the real robot arm, move_servoj and move_servo_cart **can not be used concurrently**since they use the same model(1) to control and may interfere with each other! 
```
I suppose we can't use this service with realMove_exec.launch (I tried it and it's actually interfered)

I want to visualize the robot state in Rviz while using this servo service.
Is there any launch file that streams the robot state in Rviz without bringing up the Moveit?

penglongxiang, 2022-03-21:
Hi, currently there is no such launch file that satisfies this condition. We will try to add RViz visualization when bringing up the robot without Moveit controller, as in: `$ roslaunch xarm_bringup xarm7_server.launch robot_ip:=192.168.1.128 ` for all models in next update.

If you are in a hurry, you may need to add something in the above-mentioned launch file, including loading `/robot_description`, starting `joint_state_publisher` (specify `/source_list` as `/xarm/joint_states`), `robot_state_publisher` nodes, and bringing up `RViz` with robot model added.

vimior, 2022-03-22:
@petpetpeter 
Please update to the latest xarm_ros and start the rviz display through the `show_rviz` parameter at startup.
`roslaunch xarm_bringup xarm6_server.launch robot_ip:=192.168.1.196 show_rviz:=true`

petpetpeter, 2022-03-30:
That's work for me. Thank you for your update.

closed 2022-03-30 by petpetpeter