xArm-Developer/xarm_ros #101: Servo_Cartesian Tool Frame

david-sperling-Genesis, 2022-03-24:
When following the tutorial on Servo_Cartesian https://github.com/xArm-Developer/xarm_ros/tree/master/examples, when I enter the command: `rosservice call /xarm/move_servo_cart [0,0,2,0,0,0] 0.0 0.0 1 0.0` after following the other lines, the arm tries to go to global coordinates [0,0,2,0,0,0] (stopped by E-Stop) instead of moving 2mm forward on the tool face. I have checked for typos and repeatability. I am using an xARM 7 (XS1300) with controller firmware 1.8.5 and ROS Melodic on Ubuntu 18.04.

I would like to use tool frame commands in my code, but cannot until I figure out how to implement them without the robot rushing toward home.

penglongxiang, 2022-03-25:
Hi @david-sperling-Genesis , thank you for pointing out this bug. We have just fixed it in the latest commit. Please check it out and test again.

david-sperling-Genesis, 2022-06-21:
The above-mentioned commit fixed the issue. Thank you

closed 2022-06-21 by david-sperling-Genesis