xArm-Developer/xarm_ros #102: Adding new custom ROBOTIS HAND RH-P12-RN

SOHAILMBS, 2022-04-06:
Hi, we've been trying to integrate a ROBOTIS gripper RH-P12-RN with a xARM6. Currently, it is connected with the serial port on the end of the xARM6.

We have tried communicating with the RSH-485 port but have been unsuccessful. We followed this [example](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/thridparty/set_yinshi_gripper.py) in which some similar grippers like the Yinshi and Robotiq that were controlled via the RSH-485 here but it doesn't seem to work with this gripper.

Any potential solutions or insights into integrating and using this tool would be highly appreciated.

penglongxiang, 2022-04-11:
Hi @SOHAILMBS, 
(1) have you configured with correct baudrate? 
(2) What was the sent/responded data and error output? 
(3) What was the firmware version used? 
Please provide more information to us, thank you.

rubendsa, 2022-06-01:
> Hi, we've been trying to integrate a ROBOTIS gripper RH-P12-RN with a xARM6. Currently, it is connected with the serial port on the end of the xARM6.
>
> We have tried communicating with the RSH-485 [...]

@SOHAILMBS Did you ever get this working? I'm also looking at potentially using the ROBOTIS HAND RH-P12-RN with an Xarm 7.

Cheers

SOHAILMBS, 2022-06-02:
@rubendsa Hi, we tried integrating it but it was taking a lot of time and augmentations due to which we shifted to using a robotiq-2f-140 gripper. Some information that might help you is a [tweet](https://twitter.com/k_yone/status/1431836684685955083?s=20&t=HDaNYl4JmtKlF8aoao4agQ) by a Japanese engineer who also tried to integrate the gripper courtesy of the Robotis hand devs.

closed 2022-06-02 by SOHAILMBS

rubendsa, 2022-06-04:
Thanks @SOHAILMBS , This helps a lot! Did you by any chance try changing the protocol of the ROBOTIS HAND RH-P12-RN to use Modbus instead of the default Dyanmixel 2.0 Protocol? Something like this: 
[image]
using the Dyanmixel 2.0 Wizard https://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/

Thanks again, 

-Ruben