xArm-Developer/xarm_ros #104: Xarm calibration issue

jeoseo, 2022-04-12:


penglongxiang, 2022-04-13:
Hi @jeoseo, you mean the reported position is off even in static state right? I can see from this picture that J5 may not be in the zero position. Was the robot in correct position when you first received it and did this robot have any severe collision with the environment?

jeoseo, 2022-04-13:
Yes, even in the static state. Unfortunately I cannot guarantee that the robot was in the correct position when we first got it, but I am sure we could've knocked it out of alignment as the arm has fallen over before.

MinnaZhong, 2022-04-14:
Hi Jeoseo,

The zero point of the arm seems to have changed because of the fall, I am afraid we have to get the arm back and calibrate it.
Please provide the SN of the robot to email-minna.zhong@ufactory.cc.
Please make sure the arm is in zero position, and take a picture for us.

jeoseo, 2022-04-14:
I can take a better picture tomorrow, but how long would this fix take? Sending it from here to China and back would probably make us miss a project deadline.. Is there anyway to change the zero point from here?

MinnaZhong, 2022-04-14:
Hi Jeoseo,

It's almost impossible to do the calibration by yourself, let me get a better picture and confirm it.
Normally it will take 3 weeks, including repair and shipment.

jeoseo, 2022-05-10:
This was for a university project, and we ended up completing it with a manual (kind of inaccurate) calibration. Thanks for the feedback though.

closed 2022-05-10 by jeoseo