xArm-Developer/xarm_ros #105: Issues of the model collapsing in Gazebo

hiroarkMani, 2022-04-14:
If I run "roslaunch urdf_tutorial display.launch <model>" or with the real machine ("realMove_exec.launch <IP>"), it works fine. However, when I display this URDF in gazebo, it shows the xArm in a collapsed state as shown in the image. it works in gazebo if I don't add anything to the link_base, but I want to simulate it in a real environment, so I am having trouble.
[image]

For reference, the URDFPreview is shown below.
[image]

penglongxiang, 2022-04-14:
Hi, please carefully adjust the object positions and make sure the ground, desk and xArm are not in collisions with each other. It is  recommended to leave some extra space in between, since it's easy to become unstable and out of control for collisions in Gazebo, even if there is only a small overlay of 0.00001 m.

hiroarkMani, 2022-04-14:
Hi, thanks for the quick reply, I wanted to make the base_link the world coordinate, so I placed the desk in the negative direction of the z axis, but the z=0 plane interferes with the desk in the ground determination. I placed the desk on the ground and it worked fine. Thank you .