xArm-Developer/xarm_ros #106: Quaternion to RPY for Servo_Cartesian

david-sperling-Genesis, 2022-04-19:
Hello, I am trying to make the xArm7 tele-operate with a Phantom Omni input device. The Phantom Omni specifies its rotation state as a quaternion, but the the xArm7 asks for roll, pitch, and yaw. As I looked online, I found that there are many different possible conventions that use the words roll, pitch, and yaw and want to make sure I am using the right one. I tried to implement a simple quat2rpy converter, but it is not giving me the tool motion I am expecting.

I looked through the user manual and tried to implement an extrinsic conversion that is ordered XYZ with space-fixed axes (Y & Z axes don't move when X rotation is applied, Z doesn't move when Y is applied), but it still isn't giving me the movement I would expect.

Do I have a correct understanding of the expected input to the Servo_Cart service (extrinsic, space-fixed, XYZ-ordered rotation)?

Current Quaternion to RPY Code:
```
geometry_msgs::Vector3 quat2rpy(geometry_msgs::Quaternion qi) 
{
    geometry_msgs::Vector3 rpy;
    // https://www.autonomousrobotslab.com/frame-rotations-and-representations.html
    
    // roll (x-axis rotation)
    double sinr_cosp = 2 * ((qi.w * qi.x) + (qi.y * qi.z));
    double cosr_cosp = ((qi.w*qi.w) + (qi.z*qi.z) - (qi.x*qi.x) - (qi.y*qi.y));
    rpy.x = std::atan2(sinr_cosp, cosr_cosp);

    // pitch (y-axis rotation)
    double sinp = 2 * ((qi.w * qi.y) - (qi.z * qi.x));
    if (std::abs(sinp) >= 1)
        rpy.y = std::copysign(M_PI / 2, sinp); // use 90 degrees if out of range
    else
        rpy.y = std::asin(sinp);

    // yaw (z-axis rotation)
    double siny_cosp = 2 * ((qi.w * qi.z) + (qi.x * qi.y));
    double cosy_cosp = ((qi.w * qi.w) + (qi.x * qi.x) - (qi.y*qi.y) - (qi.z * qi.z));
    rpy.z = std::atan2(siny_cosp, cosy_cosp);

    return rpy;
}    
```

penglongxiang, 2022-04-20:
Hi @david-sperling-Genesis, our roll-pitch-yaw is expressed as `X-Y-Z order extrinsic` (Fixed Coordinate) rotation. However, please make sure the formula you used is targeted for this expression, as I saw from the derivation in the reference page, it seems to be in Z-Y-X intrinsic rotation. 

If that is the case, try to **exchange the result for roll and yaw**, see if that combination is your desired result. Because it seems R_intrinsicZYX(alpha, beta, gamma) is equal to R_extrinsicXYZ(gamma, beta, alpha).

penglongxiang, 2022-04-20:
Hi @david-sperling-Genesis , your understanding is correct, our roll-pitch-yaw is expressed as X-Y-Z order extrinsic (fixed coordinate) Tait-Bryan angles. 

Refer [here](https://github.com/xArm-Developer/xarm_ros/blob/master/doc/xArmFrames.png), the three angles represent how the tool coordinate (B) is oriented with respect to base coordinate (A).

penglongxiang, 2022-04-20:
May I ask whether you're using servo_cartesian in base (default) or tool coordinate? If the problem occurs in tool coordinate, please update your local repository since we recently resolved a bug that tool coordinate command may not be effective.

david-sperling-Genesis, 2022-04-20:
Thank you for confirming the convention. I am using base-coordinates and my local repo is up-to-date.

I will continue with component testing and simplification to look for bugs elsewhere in my code.

closed 2022-04-20 by david-sperling-Genesis