xArm-Developer/xarm_ros #107: Error codes reported by the ROS driver

BryanStuurman, 2022-05-20:
The xarm manual details a number of error codes, all with letter prefixes; whereas the ROS driver reports a numeric error code only. Is there a guideline for interpreting the ROS driver numeric error codes? are they always going to align with the controller error codes (Cxx page 85 of the dev manual)?

penglongxiang, 2022-05-25:
Hi @BryanStuurman, Normally the error codes you get from "`get_err`" service or the "`err`" filed in "`xarm_states`" message are consistent with **controller error code** (Cxx codes popped up in xArm Studio). We have noticed the inconvenient situation and will try to add more interpretation to the numerical codes in next updates.

BryanStuurman, 2022-05-31:
Thanks for the information.

A question: "Normally" they coincide with Cxx error codes. under what conditions do they **not** coincide with Cxx error codes?

penglongxiang, 2022-06-01:
Since the error is reported by controller, if the communication channel has no problem, it should always coincide with Cxx code. I have not observed inconsistency on my side yet. If you have experienced the abnormal situation please report to us and we will investigate immediately, thank you.

BTW, we have already added Cxx code interpretation in the latest commit, please update if you need this function.

closed 2022-06-24 by BryanStuurman