xArm-Developer/xarm_ros #111: Velocity control feed forward gains

BryanStuurman, 2022-06-24:
For all the velocity trajectory controllers, the velocity_ff parameters are very low:
```
velocity_ff:
    joint1: 0.25
    joint2: 0.25
    joint3: 0.25
    joint4: 0.25
    joint5: 0.25
    joint6: 0.25
```

This is generally not how these controllers are set up; when one generates a trajectory the velocity portion is a perfect differentiation of the path, so in the ideal case the robot could move through the sequence of velocities and never deviate from the intended path. Here instead the feed forward term reduces the intended velocity by a factor of 4, and relies on the error correction part of the controller to make up the difference, leading to much higher than usual proportional gains and unworkable path tolerance per joint of 1 radian.

I'd like to see the velocity_ff terms raised to 1 and the error controller gains adjusted accordingly, and a path tolerance error of less than 0.1. I will be able to do some testing my one of my own 5 axis arms in about a week.

penglongxiang, 2022-06-30:
Hi @BryanStuurman , thanks for the feedback. 

Actually in our experiment we found: if fully depend on velocity feed-forward (velocity_ff:=1.0 and pid_params:=0.0 for all joints), the joint velocity command / velocity profiles received by our controller are a little bit strange, like the one shown below:

[image]

However, after tuning down the proportion of velocity feed-forward and then increase the pid (now just p) parameters, the velocity command profile received is more normal (close to triangle if max velocity not reached), although there are noises added due to pid control effect, below shows the same motion command profile with velocity_ff:=0.25 and p:=10.0:

[image]

You can adjust those parameters according to your application need, and we will be glad to know what you found during the parameter tuning.

BryanStuurman, 2022-07-01:
Those velocity trajectories are interesting to say the least, and I can see that you're using the PID controller as a bit of a low pass filter to make them more suitable for the robot to execute.

How did you generate those trajectories? Was this with moveit? The default post processor in moveit is known to be subomtimal to say the least. When we use moveit we generally use it with the TOTG post processor. Currently we generate our trajectories with the mstraj() function in roboticstoolbox by Peter Corke (https://github.com/petercorke/robotics-toolbox-python); here's a sample trajectory that we would feed forward at a gain of 1.0:
[image]

I will see about a moveit patch suggestion as well to get it generating smoother trajectories for the community.

penglongxiang, 2022-07-01:
Hi @BryanStuurman , yes the velocity profile I showed comes from moveit. After some research, as stated [here](https://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/time_parameterization/time_parameterization_tutorial.html#time-parameterization-algorithms), Moveit uses Iterative Parabolic Time Parameterization (**IPTP**) algorithm by default for time parameterization, but it has a known [bug](https://github.com/ros-planning/moveit/issues/160) that may generate abnormal oscillations after Quintic interpolation. 

However, if switch to Time-optimal Trajectory Generation(**TOTG**) algorithm, as can be configured in Moveit after ROS Melodic, or as the default used in ROS2. The velocity profile is back to normal. Check out the output of TOTG from the same motion planning task by Moveit:

[image]

In this way I guess the `velocity_ff `can be set to 1.0 and depend less on PID controller. The problem is: TOTG is not available in binary installed Moveit with ROS Kinetic, which lots of our customers are still using. For now, I guess I will keep current configurations and pin this issue for anyone who has the interest for switching algorithms and tuning parameters.

The way to **switch to TOTG with ROS Melodic or later** platform is: editing `ompl_planning_pipeline.launch.xml` in corresponding moveit_config package,  change `default_planner_request_adapters/AddTimeParameterization` to `default_planner_request_adapters/AddTimeOptimalParameterization`.

penglongxiang, 2022-07-01:
And for the other supported method - Iterative Spline Time Parameterization (ISTP), available in Kinetic, the output velocity profile for this motion can be even more abnormal:

[image]

So I would not change the configuration to use this algorithm either.

penglongxiang, 2022-07-05:
Sure, thanks for the discussion, we did not find out the root cause before.

closed 2022-07-05 by penglongxiang