xArm-Developer/xarm_ros #112: Baud rate error with Robotiq 2F-140 gripper

adamanov, 2022-07-13:
Hello, we have xArm 6dof  and attached Robotiq 2F-140 gripper on it. We recevied them as already gripper connected on controller box.

I try to to activate a gripper with already provided action server `/xarm/xarm_gripper/gripper_action/goal ` by setting a target position of gripper.  Whenever I publish on the topic I receive 
```
[ INFO] [1657727020.623691996]: gripper_action_goal handle, position: 0.500000
[ WARN] [1657727020.627824831]: set_gripper_mode, ret=22, err=0, cur_pos=0.000000
```
Which means the error is related to [modbus baudrate not correct](https://github.com/xArm-Developer/xArm-CPLUS-SDK/blob/master/doc/xarm_api_code.md#api-code). However whenever I execute a example script [/example/thirdparty-set_robotiq_gripper.cc](https://github.com/xArm-Developer/xArm-CPLUS-SDK/blob/bf47881e7457d12abcbb7b4f70a0a4ce148f6656/example/thirdparty-set_robotiq_gripper.cc), the gripper works fine (by default baud rate for[ robotiq gripper is 115200](https://github.com/xArm-Developer/xArm-CPLUS-SDK/blob/bf47881e7457d12abcbb7b4f70a0a4ce148f6656/src/xarm/wrapper/xarm_api.cc#L201) ) I dont undestand why i got baud rate error whenever I send an action goal command.  Bellow you can find my launch file. 

I would appreciate if you could help me. Thanks in advance.

```
<?xml version="1.0"?>
<launch>
    <arg name="xarm_hw_ns" default="xarm" />
    <arg name="ext_ns" default="/" />

    <arg name="robot_ip" default="192.168.132.1" />
    <arg name="report_type" default="normal" />
    <arg name="baud_checkset" default="true" />
    <arg name="default_gripper_baud" default="115200" />
    <arg name="velocity_control" default="false" />
    <arg name="enforce_limits" default="true" />
    <arg name="add_gripper" value="true" />

    <!-- ==================== Load Arm HW ====================  -->
    <!-- common: bring up node to enable real xArm control -->
    <param name="$(arg xarm_hw_ns)/uf_model" value="XARM" />
    <param name="$(arg xarm_hw_ns)/DOF" value="6" />

    <!-- common: bring up node to collect robot feedback -->
    <param name="$(arg xarm_hw_ns)/xarm_robot_ip" type="str" value="$(arg robot_ip)" />
    <param name="$(arg xarm_hw_ns)/xarm_report_type" type="str" value="$(arg report_type)" />
    <param name="$(arg xarm_hw_ns)/baud_checkset" type="bool" value="$(arg baud_checkset)" />
    <param name="$(arg xarm_hw_ns)/default_gripper_baud" type="int" value="$(arg default_gripper_baud)" />
    <param name="$(arg xarm_hw_ns)/velocity_control" type="bool" value="$(arg velocity_control)" />
    <param name="$(arg xarm_hw_ns)/enforce_limits" type="bool" value="$(arg enforce_limits)" />
    <param name="$(arg xarm_hw_ns)/add_gripper" type="bool" value="$(arg add_gripper)" />

    <node name="uf_ros_controller" pkg="xarm_controller" type="uf_ros_controller" args="ip=$(arg robot_ip)" ns="$(arg xarm_hw_ns)" respawn="false" output="screen" />

</launch>
```

penglongxiang, 2022-07-15:
Hi @adamanov, the provided xarm_gripper/gripper_action is targeted for the gripper from UFACTROY, the underlying communication protocol is customized and requires a baudrate of 2M. You may need to write another action server for Robotiq gripper.

vimior, 2022-07-18:
Hi @adamanov 
1. At present, the gripper_action in ros is only for the control and feedback of xarm_gripper
2. The different peripherals set by set_checkset_default_baud are the default values of the baud rate to be checked (type corresponds to the peripheral type, and different types are only valid for different APIs), because gripper_action uses xarm_gripper-related APIs, so even if set_checkset_default_baud (3, 115200) is also useless. The APIs related to robotiq should be similar to robotiq_xxx, but these APIs are only in the SDK and not provided in ros.
3. If you want to control robotiq through gripper_action without changing the code of xarm_api, you can customize gripper_action and communicate with robotiq through the set_tool_modbus service in xarm_api

adamanov, 2022-07-26:
> Hi @adamanov
>
>     1. At present, the gripper_action in ros is only for the control and feedback of xarm_gripper
>
>     2. The different peripherals set by set_checkset_default_baud are the default [...]

Hi  @vimior 
Thank you very much for your detailed answer, and sorry for late response. 
I fixed the the problem by writing my own Gripper Action Service which communicate with Gripper Hardware over xArm_API and also published a joint states of gripper for Moveit in order to send commands over Rviz Moviet.

closed 2022-07-26 by adamanov