xArm-Developer/xarm_ros #113: Trajectory recording and Playing back for XARM

mbs-tahir, 2022-08-12:
I would like to ask you is about Trajectory recording and playback on arm. I have tested the provided scripts on GitHub in [xarm_python_sdk](https://github.com/xArm-Developer/xArm-Python-SDK) for [recording](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/3002-record_trajectory.py) and [playing back](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/example/wrapper/common/3003-playback_trajectory.py) trajectories. But I can't find how to record trajectory in xarm_ros though I can see set_trajectory, save_traj, play_traj in ROS services but don't know how to record one. Any leads on that will highly be appreciated.

mbs-tahir, 2022-08-12:
I have a few more questions about the trajectory recorded and access to the source `.traj` file.

1. As per documentation in in the [SDK](https://github.com/xArm-Developer/xArm-Python-SDK/blob/master/xarm/wrapper/xarm_api.py#L1149), trajectories are saved in the controller box. Is there any way to get access to that or copy them from controller box to the comupter and vice versa? This will be helpful in visualizing trajectories in RVIZ. 
2. For how long the trajectories saved are stored in the controller box? Also is it possible to see list of available `.traj` files on the controller box?

vimior, 2022-08-15:
@mbs-tahir 
1. Since the recording and playback of the track is 250HZ, the recording and playback of the track are currently controlled by the control box software, and there is no API access for the time being.
2. The maximum recording time is 5 minutes and 15 seconds, the same name will be overwritten