xArm-Developer/xarm_ros #118: Error Code: 11 Msg: Unable to find uri[model://table]

manny1185, 2022-08-24:
I am new to Gazebo.
When i am running "roslaunch xarm_gazebo xarm6_beside_table.launch add_vacuum_gripper:=true"

i met some issues:

process[gazebo-1]: started with pid [3642]
process[gazebo_gui-2]: started with pid [3647]
process[spawn_gazebo_model-3]: started with pid [3651]
process[xarm/traj_controller_spawner-4]: started with pid [3653]
process[xarm/gripper_controller_spawner-5]: started with pid [3654]
process[robot_state_publisher-6]: started with pid [3655]
[INFO] [1661348672.061392, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1661348672.628837659]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1661348672.632292646]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1661348672.652047716]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1661348672.655733161]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1661348672.930134, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1661348672.936311, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
Warning [parser.cc:759] XML Attribute[update_rate] in element[physics] not defined in SDF, ignoring.
Error Code: 11 Msg: Unable to find uri[model://table]
[ INFO] [1661348676.663288605]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1661348676.697167730, 0.010000000]: Physics dynamic reconfigure ready.
[INFO] [1661348676.897872, 0.181000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1661348677.264227, 0.359000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1661348677.438110555, 0.359000000]: Loading gazebo_ros_control plugin
[ INFO] [1661348677.440594757, 0.359000000]: Starting gazebo_ros_control plugin in namespace: /xarm
[ INFO] [1661348677.442130997, 0.359000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[spawn_gazebo_model-3] process has finished cleanly
log file: /home/manny/.ros/log/99b941de-23a7-11ed-a1ea-000c29522bd0/spawn_gazebo_model-3*.log
[ERROR] [1661348677.562069069, 0.359000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint1
[ERROR] [1661348677.563545699, 0.359000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint2
[ERROR] [1661348677.564666350, 0.359000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint3
[ERROR] [1661348677.565901515, 0.359000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint4
[ERROR] [1661348677.566954192, 0.359000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint5
[ERROR] [1661348677.568010346, 0.359000000]: No p gain specified for pid.  Namespace: /xarm/gazebo_ros_control/pid_gains/joint6
[ INFO] [1661348677.584219157, 0.359000000]: Loaded gazebo_ros_control.
[INFO] [1661348677.804410, 0.541000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1661348677.808402, 0.541000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1661348677.812018, 0.542000]: Loading controller: gripper_trajectory_controller
Loaded 'xarm6_traj_controller'
[ERROR] [1661348677.957237679, 0.632000000]: Could not find joint 'drive_joint' in URDF model.
[ERROR] [1661348677.957440520, 0.632000000]: Failed to initialize the controller
[ERROR] [1661348677.957877243, 0.632000000]: Initializing controller 'gripper_trajectory_controller' failed
Loaded 'joint_state_controller'
Started ['xarm6_traj_controller'] successfully
Started ['joint_state_controller'] successfully
[xarm/traj_controller_spawner-4] process has finished cleanly
log file: /home/manny/.ros/log/99b941de-23a7-11ed-a1ea-000c29522bd0/xarm-traj_controller_spawner-4*.log
[Err] [REST.cc:205] Error in REST request

libcurl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to api.ignitionrobotics.org:443 
[ERROR] [1661348678.958659, 1.461000]: Failed to load gripper_trajectory_controller
[INFO] [1661348678.959733, 1.461000]: Controller Spawner: Loaded controllers: 
[INFO] [1661348678.964260, 1.467000]: Started controllers: 

how can i slove it?

penglongxiang, 2022-08-25:
Hi, please make sure you have finished all the preparation work in ReadMe chapter 3 correctly. Refer: https://github.com/xArm-Developer/xarm_ros#33-download-the-table-3d-model

manny1185, 2022-08-25:
> Hi, please make sure you have finished all the preparation work in ReadMe chapter 3 correctly. Refer: https://github.com/xArm-Developer/xarm_ros#33-download-the-table-3d-model

I thought it wasn't required so I skipped this step, sorry

closed 2022-08-25 by manny1185