xArm-Developer/xarm_ros #119: Lite6 planner available?

sean880304, 2022-09-01:
Hi, 

I follow the xArm Simple Planner instruction and try to launch it as a Lite6 model by changing the namespace but didn't work.

Does it currently support Lite6 model? If not, is there any other way to control Lite6  in moveit! by giving a specified target point?

Thank you very much!!

sean880304, 2022-09-03:
Update:

I tried this command to launch lite6 planner:
`
roslaunch xarm_planner xarm_planner_rviz_sim.launch robot_dof:=6 robot_type:=lite
`
I think this is the correct way to do it but still couldn't activate the planner service.

`
[FATAL] [1662202619.167941723]: Group 'xarm6' was not found.
terminate called after throwing an instance of 'std::runtime_error'
`

`
what():  Group 'xarm6' was not found.
`

`
[xarm_move_group_planner-6] process has died [pid 14590, exit code -6, cmd /home/sean/catkin_ws/devel/lib/xarm_planner/xarm_simple_planner /DOF:=/xarm/DOF __name:=xarm_move_group_planner __log:=/home/sean/.ros/log/212c27a4-2b77-11ed-9f45-073ce1e45d6f/xarm_move_group_planner-6.log].
log file: /home/sean/.ros/log/212c27a4-2b77-11ed-9f45-073ce1e45d6f/xarm_move_group_planner-6*.log
`

vimior, 2022-09-05:
Hi, @sean880304 
Sorry, this package is not compatible with lite6, I temporarily compatible, please update to the latest code and try again.

sean880304, 2022-09-05:
Hi, @vimior 
Thank you very much, it works now!!