xArm-Developer/xarm_ros #121: How to set the arm pose

jzx-gooner, 2022-09-05:
1.There is a plane in front of the robot arm. I want my tool to be perpendicular to the plane
2.I fit the plane through the point cloud and transfer it to the base coordinate system
3.The normal vector of the plane is obtained
4.Through the normal vector, how can I get the pose of the robot arm so that my tool is perpendicular to the plane

thank you！

penglongxiang, 2022-09-05:
Hi @jzx-gooner, although this is not an issue of our package, I can give some basic suggestion: you may need to get the normal **unit vector** of the plane expressed **in robot base coordinate** (or the external user coordinate if you have set it for xArm controller), if normal vector is pointing outwards, it should be set as the negative Z axis of xArm TCP frame (denote this unit vector as _**z0**_).  
Then find two perpendicular unit vectors (expressed also in robot base coordinate) on the plane as target X and Y axis for xArm TCP (denote as **_x0_**, **_y0_**). The target TCP rotation matrix can be formed as **R** = [**_x0, y0, z0_**], finally transform this rotation matrix to **Roll-Pitch-Yaw** (refer to user manual for the definition) and then give it for execution.

jzx-gooner, 2022-09-07:
> Hi @jzx-gooner, although this is not an issue of our package, I can give some basic suggestion: you may need to get the normal **unit vector** of the plane expressed **in robot base coordinate** (or [...]

Thakn you very much！！！
I follow your advices and make it work
I will close this issue and Best wishes！

closed 2022-09-07 by jzx-gooner