xArm-Developer/xarm_ros #123: rosservice call /xarm/set_tool_modbus convention

david-sperling-Genesis, 2022-09-09:
I am trying to add a custom EOAT to the xArm7 that communicates over the built-in RS485 A+B wires. I would like to use the `/xarm/set_tool_modbus` service to send new position commands to an arduino (code at bottom) with a RS485 hat. The setup is as follows:
[image]
I am struggling with the convention for the data packet. For the sake of the issue, I have simplified the use to remove the servo.

Based on the [Modbus wikipedia page](https://en.wikipedia.org/wiki/Modbus) and verified on an [online parser](https://rapidscada.net/modbus/ModbusParser.aspx), if I want to read the first register, I should send a message of the following form:
`slave address, function code, starting address (16-bit), quantity (16-bit), CRC (16-bit)`
example:
`01 03 00 00 00 01 84 0a` (hex)

I can send this message to the arduino using a USB to RS485 adapter and [QModbus](http://qmodbus.sourceforge.net/) and I receive the appropriate response: 
`01 03 02 00 64 b9 af` (hex)

When I transition to the xArm service and try to send the same command (removing CRC), I receive the following response:
```
Terminal 1:
robotics@robotics:~$ roslaunch xarm_planner xarm_planner_realHW.launch robot_ip:=192.168.1.206 robot_dof:=7
Terminal 2:
robotics@robotics:~$ rosservice call /xarm/config_tool_modbus 9600 20
message: "set_modbus_baudrate, ret=0 | set_modbus_timeout, ret=0"
ret: 0
robotics@robotics:~$ rosservice call /xarm/get_tgpio_modbus_baudrate
data: 9600
ret: 0
message: "ret = 0, current baud_rate = 9600"
robotics@robotics:~$ rosservice call /xarm/set_tool_modbus [0x01,0x03,0x00,0x00,0x00,0x01] 5
ret: 0
message: ''
respond_data: [70, 3, 160, 92, 127]
```
I also receive error C19 [ End Module Communication Error ]

I also tried sending the same command (with CRC) and got the same error and `respond_data: [14, 0, 16, 19, 127]`.

Since it works with QModBus and the USB adapter, I think my arduino code works. Now I am just confused as to how I use the `/xarm/set_tool_modbus` service to send the same command. I am also interested in function 0x10 (Write Multiple Holding Registers), but I wanted to simplify the question to a single function.

How can I use the `/xarm/set_tool_modbus` to replace my QModBus+Adapter solution?

Arduino Code:
```
#include<ArduinoRS485.h>
#include<ModbusRtu.h> //Library for using Modbus in Arduino

Modbus bus;
uint16_t modbus_array[] = {100,255}; //Initialize Modbus array

void setup() {
  Serial1.begin(9600); // Modbus slave baudrate at 9600
  Serial.begin(115200); // Communication to Arduino IDE Serial Monitor
  bus = Modbus(1,Serial1,8); //Modbus slave ID 1, TX/RX for Serial1, DE/RE on pin 8
  bus.start();

}

void loop() {
  bus.poll(modbus_array,sizeof(modbus_array)/sizeof(modbus_array[0])); //Used to receive or write value from Master   

  //Print current array to serial
  for (size_t i = 0; i < sizeof(modbus_array)/sizeof(modbus_array[0]); i++)  {
    Serial.print(String(modbus_array[i]));
    Serial.print(", ");
  }
  Serial.print("\n");
}
```

vimior, 2022-09-13:
Hi, @david-sperling-Genesis 
You can update the code to the latest, and then try this service (__set_tgpio_modbus_timeout__/__getset_tgpio_modbus_data__) to see if it can meet your needs.
```
# set timeout
rosservice call /xarm/set_tgpio_modbus_timeout 20 true

# transparent transmission modbus data
rosservice call /xarm/getset_tgpio_modbus_data  [0x01,0x03,0x00,0x00,0x00,0x01] 5 9 true false
```

vimior, 2022-09-13:
Hi, @david-sperling-Genesis 
__rosservice call /xarm/getset_tgpio_modbus_data  `send_data`  `recv_len` `host_id` `transparent transmission or not`  `use port 503 or not`__
If you specify the use of transparent transmission, you need to add CRC by yourself (if necessary)