xArm-Developer/xarm_ros #127: moveit_config warning

HyeWon33, 2022-09-21:
Hello, I'm running xArm5.
I wonder why there is a warning when I use the command below to operate the real robot.

~~~
$ roslaunch xarm_bringup xarm5_server.launch robot_ip:=my_robot_ip
~~~

```
$ roslaunch xarm5_gripper_moveit_config realMove_exec.launch robot_ip:=my_robot_ip
```

The warning message is as follows.

~~~
[ WARN] [1663735467.596881460]: processStatus: Previously received status from [/xarm/uf_driver], but we now received status from [/xarm/uf_ros_controller]. Did the ActionServer change?
~~~

So what I'm curious about is

1. Is it right to use the above command to operate the actual robot?
2. Why does the above error message come out?
3. If you look at xarm5_server.launch in xarm_bringup/launch, they tell you to use xarm*_moveit_config package if you use MoveIt, what do you want me to do?
4. What I want to do is I want to run a real robot with the rivz, and I want to run it with the code I made. What command should I execute to do this?

penglongxiang, 2022-09-22:
Hi @HyeWon33 

1. the two launch methods are not supposed to be used at the same time. The first is for controlling with our ros wrapper SDK services, but currently it doesn't launch rviz with loaded robot model. The second is used for controlling with Moveit visualized by rviz.

2. This message pops up might because both the /xarm/uf_driver (used by previous launch) and /xarm/uf_ros_controller (used by second launch) will publish to the /xarm/joint_states topic, which serves as feedback source of FollowJointTrajectory Action used by Moveit Execution part.

3&4. This depends on how you would like to control the robot, could you elaborate on what interface your code uses for robot command?

HyeWon33, 2022-09-22:
I use Moveit for robot command.

penglongxiang, 2022-09-22:
Then just the second launch you mentioned should be enough.

HyeWon33, 2022-09-22:
Then how do I control a real robot using rivz?

penglongxiang, 2022-09-22:
After the launch, rviz will automatically pop up, an interactive marker will appear at the robot tool end, drag the arrows to the desired position, click "plan" and then "execute" in planning tab if trajectory is OK, the real arm will then execute the planning from Moveit.

HyeWon33, 2022-09-23:
I ran Moveit, but I got the same error as in the picture.
Commands Executed
```
$ roslaunch xarm5_gripper_moveit_config realMove_exec.launch robot_ip:=my_robot_ip
```
Error picture
[image]

There is an error called C31 at the terminal over there, so I read the manual, but I don't know what to do.

I want to know why this error occurred and how to fix it.

penglongxiang, 2022-09-28:
This is due to abnormal current sensed at the joints when collision detection is enabled. Please make sure the tool payload mass, center of mass are set properly. If robot is not mounted on a horizontal surface, the mounting method should also be configured. 

If above parameters are set but error still persists, try reducing the collision detection sensitivity or execute with smaller acceleration and  speed.

HyeWon33, 2022-10-03:
Thank you for letting me know. I solved the problem, but another error came out.
[image]
I just executed the command below.
```
$ roslaunch xarm5_gripper_moveit_config realMove_exec.launch robot_ip:=my_robot_ip
```

Akumar201, 2022-10-04:
I would suggest you to set the collision sensitivity to default and do some testing, as this happens when the arm motors gets a disturbed feedback due to striking or jerk.

HyeWon33, 2022-10-13:
Thank you for your answer. But what should I do about it?

Akumar201, 2022-10-13:
In order to remove it you can use uFactory application and move it to initial position and then try running it using moveit. I would also recommend you to first plan and see if there is any abnormal motion happening like collision or stiking with anyother  object including ground. Also check degree of freedom that you are providing to move the arm as default is 7. You can use below code to use it

`roslaunch xarm7_moveit_config realMove_exec.launch robot_ip:=<your controller box LAN IP address> [velocity_control:=false] [report_type:=normal] `
Also see in the error that it states something about mode so check which mode is currently on what it correspond to.