xArm-Developer/xarm_ros #129: Arm stiked the ground while using moveit

Akumar201, 2022-09-26:
Hello , I was following the github tutorial using the command to move the arm using Moveit. However , sometimes it strikes the ground. 
I am using this code 

$ roslaunch xarm7_moveit_config realMove_exec.launch robot_ip:=<your controller box LAN IP address> [velocity_control:=false] [report_type:=normal]

penglongxiang, 2022-09-27:
Hi, could you please provide more information so that we can reproduce on our side? What command did you send to cause the strike? Is there any error code reported by controller?

Akumar201, 2022-09-27:
Yes, sure 
I am using this command to operate arm in real using ,moveit. it is planning path where it is going under the floor. 

[image]

penglongxiang, 2022-09-28:
... So is there anything wrong? You mean the robot hits the ground because it is commanded to do so (go below ground) ?

Akumar201, 2022-09-28:
Greetings, Yes It hits the ground, however it should not even in moveit, is there some wrong parameter I am using on moveit or something is wrong with moveit configuration file. In short I would like to know how the robot will know that it can't go beyond the ground level in moveit. 
Thank you

penglongxiang, 2022-09-29:
In that case you should probably add collision objects as environment configuration to the planning scene. Please check on moveit tutorials.

Akumar201, 2022-09-29:
Should I create new moveit package or should I edit the given one ?

Akumar201, 2022-10-26:
This issue can be resolved by two methods, 
1. The easiest way would be to add a box underneath the robot in the URDF that your moveit config is based on.  You could add something like this to your URDF, 

```
<link name="ground_plane_box">
    <visual>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
            <box size="1.0 1.0 0.5"/> 
        </geometry>
        <material name="Grey">
            <color rgba="0.7 0.7 0.7 1.0"/>
        </material>
    </visual>
    <collision>
        <origin xyz="0 0 0" rpy="0 0 0"/>
        <geometry>
            <box size="1.0 1.0 0.5"/> 
        </geometry>
    </collision>
</link>

<joint name="ground_plane_box" type="fixed">
    <parent link="world" />
    <child link="ground_plane_box" />
    <origin xyz="0 0 -0.25" rpy="0 0 0" />
</joint>
```

2. The second method is to use code to include the object, please visit Adding Objects to the Planning Scene section in https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html#the-entire-code

closed 2022-10-26 by Akumar201