xArm-Developer/xarm_ros #130: rqt_Easy_handeye_Hand-eye not working

Akumar201, 2022-09-27:
Greetings, 
I am doing camera calibration using the github tutorial, using the command 

However I am not able to use rqt_Easy_handeye_Hand-eye  for changing the pose so that I can take a sample to do eye in hand calibration. Do I have to move the arm in manual mode to find the transform ? 

[image]

penglongxiang, 2022-09-28:
Please try placing the marker in a position facing the camera directly and easy for the arm tool end to change orienation. You can also consult "easy_handeye" developers for further calibration issues, we are not experts.

Akumar201, 2022-09-28:
I would like to know is something wrong with the movit package, why I am getting this warning. Thank you in advance. 

[image]

penglongxiang, 2022-09-29:
Hi, could you please provide more information, Did this warning occur only by d435i_xarm_auto_calib.launch? From my side there are no those extra camera links, have you changed anything? Thanks

Akumar201, 2022-09-29:
Greetings, 
I have not changes anything. I tried running the commands like 
$ roslaunch xarm7_moveit_config realMove_exec.launch robot_ip:=<your controller box LAN IP address> [velocity_control:=false] [report_type:=normal]
but there was no error, this error is occurring only when d435i_xarm_auto_calib.launch?
He was also facing the same issue https://github.com/IFL-CAMP/easy_handeye/issues/98

Akumar201, 2022-10-03:
Greetings, 
The error was because of realsense camera. I uninstalled the SDK for realsense, got rid of any  error that were coming during sudo apt update or sudo apt upgrade. Reinstall the SDK from https://github.com/IntelRealSense/realsense-ros.

closed 2022-10-03 by Akumar201