xArm-Developer/xarm_ros #131: (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

Akumar201, 2022-09-27:
Hello, 
I was doing camera calibration following the gihub demo. However I am getting the following error. I am using extended USB 3 cable for connection. 

[image]

penglongxiang, 2022-09-28:
This is clearly realsense related issue, please look for answers in corresponding github page [like this one](https://github.com/IntelRealSense/realsense-ros/issues/2149)  or consult the relevant developers. Thank you.

Akumar201, 2022-09-28:
To help someone who is going to encounter this error. These error's are regarding connectivity issue, try switching to original cable to troubleshoot it. If you are not getting control_transfer returned error stream continuously you can safely ignore it as the realsense takes up time to establish stable connection with the computer else if it is continuous than check or replace your cable as your connection might be dropping in between causing loss of data.

closed 2022-09-28 by Akumar201