xArm-Developer/xarm_ros #132: How to make sure that camera on base calibration is correct

Akumar201, 2022-09-29:
Hello, I am doing camera on base by setting <arg name="eye_on_hand" value="false" /> as false in calibration launch file. How can I make sure that the calibration I am doing is correct ? 

translation: 
  x: 0.4705739455942296
  y: 1.0399431884657737
  z: 0.6388135790952124
rotation: 
  x: -0.11958812174791901
  y: -0.7127805157974599
  z: 0.6887959517916398
  w: 0.056592881458372546

[image]

penglongxiang, 2022-10-09:
I am afraid you have to do some measurement from the robot base to the camera lens, to know the rough relative coordinated distance from robot base (**/link_base**) to the RGB lens (**/camera_color_optical_frame**). Then to verify with the calibrated result. 

To verify the orientation result, I made a illustration of frame orientation on the realsense d435i, just for your reference.  

[image]

image source: https://www.intelrealsense.com/depth-camera-d435i/

Akumar201, 2022-10-11:
I would like to share the  tf to verify if I am doing it correctly, please verify 
[frames.pdf](https://github.com/xArm-Developer/xarm_ros/files/9758328/frames.pdf)

penglongxiang, 2022-10-12:
hi, I am not expert on vision and calibration either, the only way I can think of is comparing with the known ground truth, and this verification needs to be done only once to confirm the reliability. Your calibration target in "eye-on-base" scenario is to calculate relative position and orientation between **robot base frame** (coordinate system) and the `/camera_color_optical frame` on the realsense camera. So just fix them at a known relative position like 2 meters away only in x direction then check the calibration result. The result given in output yaml has the units of meters and radians.

Akumar201, 2022-10-12:
I have two follow up question just to make sure things are working perfectly, so I calibrated the camera with the following result 
parameters:
  ```
eye_on_hand: false
  freehand_robot_movement: false
  move_group: xarm6
  move_group_namespace: /
  namespace: /xarm_realsense_handeyecalibration_eye_on_base/
  robot_base_frame: link_base
  robot_effector_frame: link_eef
  tracking_base_frame: camera_color_optical_frame
  tracking_marker_frame: camera_marker
transformation:
  qw: 0.053744796643656396
  qx: -0.06921099152218314
  qy: -0.7859133605577909
  qz: 0.6120960097754341
  x: 0.13260120526246572
  y: 1.4612716251100917
  z: 1.1730637467290128
```

Please let me know if I am thinking right 
  x: 0.13260120526246572 in meters is the distance from robot base frame (0,0,0) to /camera_color_optical frame in x direction
  y: 1.4612716251100917 in meters is the distance from robot base frame (0,0,0) to /camera_color_optical frame in y direction
  z: 1.1730637467290128 in meters is the distance from robot base frame (0,0,0) to /camera_color_optical frame in z direction

Akumar201, 2022-10-13:
I resolved this issue, 
So, treat the robot base as (0,0,0) , remember front is roll, side is pitch and up is yaw. Front can be defined as the side which is facing the camera. Also find out the right coordinate system for your camera for my case it was d435

[image]
where green is y, red is x and z front axis in blue. Then try to rotate the base frame by any of roll, pitch or yaw to get the same orientation as the camera. Then note down the angles for roll, pitch and yaw that can be compared to the result in easy hand-eye calibration. The quaternions in the yaml are in radians, you can use the following code to convert them into euler form in degree

```
import math
 
def euler_from_quaternion(x, y, z, w):
        """
        Convert a quaternion into euler angles (roll, pitch, yaw)
        roll is rotation around x in radians (counterclockwise)
        pitch is rotation around y in radians (counterclockwise)
        yaw is rotation around z in radians (counterclockwise)
        """
        t0 = +2.0 * (w * x + y * z)
        t1 = +1.0 - 2.0 * (x * x + y * y)
        roll_x = math.atan2(t0, t1)
     
        t2 = +2.0 * (w * y - z * x)
        t2 = +1.0 if t2 > +1.0 else t2
        t2 = -1.0 if t2 < -1.0 else t2
        pitch_y = math.asin(t2)
     
        t3 = +2.0 * (w * z + x * y)
        t4 = +1.0 - 2.0 * (y * y + z * z)
        yaw_z = math.atan2(t3, t4)
     
        return roll_x, pitch_y, yaw_z # in radians # roll and pitch can be diffrent in your case so choose respectively

val = euler_from_quaternion(-0.06259327310419002,-0.7003283367747712,0.7080282847987625,0.06571339891764572)
degrees = []
for i in range(len(val)):
    degrees.append(math.degrees(val[i]))
print(val)
print(degrees)
```

closed 2022-10-13 by Akumar201