xArm-Developer/xarm_ros #135: No transform from[camera_color_frame] to frame world

Akumar201, 2022-10-04:
I was calibrating the eye on base camera as mentioned in the tutorial. I think something is wrong as you can see camera marker is pointing somewhere else and base is pointing somewhere else also in moveit window. 
I am using Ubunut 20.04, ROS noetic and Realsense D435

[image]

It also gives an warning No transform from [camera_color_frame] to frame[world] 

[image]

The same warning is coming when doing camera in hand 

[image]

penglongxiang, 2022-10-08:
Hi @Akumar201 , today I reproduced this issue in **ROS Noetic** environment. And it seems the warnings can be removed by giving an external argument to `realsense2_camera/launch/rs_camera.launch`. Please go to xarm_vision/d435i_xarm_setup/launch/d435i_auto_calib.launch, and modify the part
```html
<!-- start the Realsense435 -->
<include file="$(find realsense2_camera)/launch/rs_camera.launch" />
```
as:
```html
<!-- start the Realsense435 -->
<include file="$(find realsense2_camera)/launch/rs_camera.launch" >
    <arg name="publish_tf" value="false" />
</include>
```

Another issue I encountered in noetic was: I have to upgrade the `ros-noetic-moveit-*` packages to the latest **1.1.10** version (released 20220927), or the checking starting position stage will always fail and output the message "**Can't calibrate from this position!**". You can do the upgrade by command: `sudo apt-get --only-upgrade install ros-noetic-moveit*` if needed.

Akumar201, 2022-10-11:
Hello, @penglongxiang  thank you for your response. The error  No transform from [camera_color_frame] to frame[world] has been resolved using the steps mentioned by you. However I am getting the following warning, can you please help me with it ?
[image]
In addition this I tried `sudo apt-get --only-upgrade install ros-noetic-moveit*` however still facing the same issue.

penglongxiang, 2022-10-12:
I have no experience of "eye-on-base" scenario, but I guess you can ignore this mismatch and finish the calibration process. Because the relative position between marker and tool-end is not known, that is, the program does not know how you fix the marker on the tool-end and it is probably also calculated from the calibration data in the end.

Akumar201, 2022-10-12:
I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2022-10-12 by Akumar201