xArm-Developer/xarm_ros #143: Moveit Api instead of Xarm Api for picka and place

arahami, 2022-11-01:
hey guys. thanks for the great API. I'm trying to reuse my existing code for another project in which I used Moveit API for pick and place for another project that uses Xarm robots. the problem is that all parts of picking-placing work except for the gripper's joints don't move and so grasping fails. is there any way to use the Grasp message with Pick and Place methods from Moveit?

penglongxiang, 2022-11-02:
Hi, our integration with moveit grasp or pick and place is not yet implemented. We will look into it and try to develop ASAP. Will get back to you once we have the answer to your question. Thanks.

vimior, 2022-11-09:
@arahami 
Hi, I tried to use moveit's pick and place, it works fine.
The tested code example is as follows (the parameter values are slightly modified based on the official example)
```C++
#include <ros/ros.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/move_group_interface/move_group_interface.h>
#include <tf2_geometry_msgs/tf2_geometry_msgs.h>

const double tau = 2 * M_PI;

void openGripper(trajectory_msgs::JointTrajectory& posture)
{
  posture.joint_names.resize(1);
  posture.joint_names[0] = "drive_joint";

  posture.points.resize(1);
  posture.points[0].positions.resize(1);
  posture.points[0].positions[0] = 0.0;
  posture.points[0].time_from_start = ros::Duration(0.5);
}

void closedGripper(trajectory_msgs::JointTrajectory& posture)
{
  posture.joint_names.resize(1);
  posture.joint_names[0] = "drive_joint";

  posture.points.resize(1);
  posture.points[0].positions.resize(1);
  posture.points[0].positions[0] = 0.75;  // max: 0.85
  posture.points[0].time_from_start = ros::Duration(0.5);
}

void pick(moveit::planning_interface::MoveGroupInterface& move_group, int cnts = 1)
{
  std::vector<moveit_msgs::Grasp> grasps;
  grasps.resize(1);

  grasps[0].grasp_pose.header.frame_id = "link_base";
  tf2::Quaternion orientation;
  orientation.setRPY(-tau / 4, -tau / 4, -tau / 4);
  grasps[0].grasp_pose.pose.orientation = tf2::toMsg(orientation);
  grasps[0].grasp_pose.pose.position.x = 0.515;
  grasps[0].grasp_pose.pose.position.y = 0;
  grasps[0].grasp_pose.pose.position.z = 0.5;

  // Setting pre-grasp approach
  grasps[0].pre_grasp_approach.direction.header.frame_id = "link_base";
  /* Direction is set as positive x axis */
  grasps[0].pre_grasp_approach.direction.vector.x = 1.0;
  grasps[0].pre_grasp_approach.min_distance = 0.095;
  grasps[0].pre_grasp_approach.desired_distance = 0.115;

  // Setting post-grasp retreat
  grasps[0].post_grasp_retreat.direction.header.frame_id = "link_base";
  /* Direction is set as positive z axis */
  grasps[0].post_grasp_retreat.direction.vector.z = 1.0;
  grasps[0].post_grasp_retreat.min_distance = 0.1;
  grasps[0].post_grasp_retreat.desired_distance = 0.25;

  // Setting posture of eef before grasp
  openGripper(grasps[0].pre_grasp_posture);

  // Setting posture of eef during grasp
  closedGripper(grasps[0].grasp_posture);

  // Set support surface as table1.
  move_group.setSupportSurfaceName("table1");
  // Call pick to pick up the object using the grasps given
  auto code = move_group.pick("object", grasps);
[... 129 lines cut ...]
  ros::init(argc, argv, "xarm_pick_place");
  ros::NodeHandle nh;
  ros::AsyncSpinner spinner(1);
  spinner.start();

  ros::WallDuration(1.0).sleep();
  moveit::planning_interface::PlanningSceneInterface planning_scene_interface;
  moveit::planning_interface::MoveGroupInterface group("xarm6");
  group.setPlanningTime(60.0);

  addCollisionObjects(planning_scene_interface);

  // Wait a bit for ROS things to initialize
  ros::WallDuration(1.0).sleep();

  pick(group);

  ros::WallDuration(1.0).sleep();

  place(group);

  ros::waitForShutdown();
  return 0;
}
```

arahami, 2022-11-10:
@vimior sorry for the late reply. thanks very much for the support. I haven't had time to test it yet. but I had a similar code that didn't work (gripper) and that was why I came to the conclusion that something was wrong. if this code is working, I only can conclude that I missed something. 
I'll double-check everything, compare my code with this one and get back to you.

arahami, 2022-11-12:
@vimior I tested the code. It works perfectly. I did compare it with my code and it turned out the problem was `posture.joint_names[0] = "drive_joint"`, whereas i used `left_finger_joint` and `right_finger_joint` similar to Moveit Tutorial. 
as a workaround moved the gripper's fingers like this:
``` python
hand_move_group = MoveGroupCommander(name="xarm_gripper")
current_joints_values: List[float] = hand_move_group.get_current_joint_values()
        new_joint_values = current_joints_values.copy()
        new_joint_values[0] = 0.5
        result = hand_move_group.go(joints=new_joint_values, wait=True)
```
now i understand that i was actually setting `drive_joint` value not `left_finger_joint` and `right_finger_joint.
thanks again for the help. this made the pick_place part much easier and now I can reuse most of the existing code.

closed 2022-11-12 by arahami