xArm-Developer/xarm_ros #145: Failed to control xarm6 real robot arm

AtariaLee, 2022-11-09:
When I connected the real robot arm, I successfully connected it, but the model in RVIZ is not the same as the real robot arm. 
The robot arm in rviz is in the default position, but it is not in reality.
And when I use moveit to plan the trajectory, errors will occur.like
`[1667995865.456842619]: ABORTED: Solution found but controller failed during execution`
The simulation manipulator running the demo can be planned.
My command to connect the real mechanical arm is:`roslaunch xarm6_moveit_config realMove_exec.launch robot_ip:=192.168.1.214`
What's the problem? How can I solve it?

penglongxiang, 2022-11-10:
Hi, if model in RVIZ is not at the same configuration with real xArm, please check if the hardware is properly powered on (emergency stop button is not effective), and if there is any error/warning message reported in terminal output (before the content you pasted  here). 

Or you can use UFACTORY Studio to make sure the robot arm can be controlled without any issue.

AtariaLee, 2022-11-10:
即停按钮没有按下，并且当我连接真实机器人的时候，机器人会发出一两下响声。现在在原先问题基础上又出现一个新的问题，
```
process[robot_state_publisher-4]: started with pid [7029]
[ WARN] [1668046062.004109896]: Shutdown request received.
[ WARN] [1668046062.004789410]: Reason given for shutdown: [[/robot_state_publisher] Reason: new node registered with same name]
```
rviz中这次不显示模型。[/home/lee/Desktop/1.png](url)

penglongxiang, 2022-11-10:
请问执行realMove_exec.launch的时候，有没有同时在跑其他程序？比如demo.launch。这样会导致一些同名节点出现冲突而出现问题。 如果并没有同时跑其他程序，可能是之前的节点没有kill掉，这时候可以重启PC试一下。但建议先用studio控制一下手臂确认一切正常没有报错。

AtariaLee, 2022-11-10:
确实有一个其他的程序在跑，关掉后现在又回到了开始的问题。

AtariaLee, 2022-11-10:
使用uf studio后有一个报错，`servo error code s22 id3`
`encode configuration error`

AtariaLee, 2022-11-10:
```
Servo Error
Code S22  ID[3]
Encode Configuration Error
```

penglongxiang, 2022-11-10:
了解，我们技术支持同事会通过邮件联系您

closed 2022-11-10 by AtariaLee