xArm-Developer/xarm_ros #146: error using"  roslaunch d435i_xarm_setup d435i_xarm_auto_calib.launch" to calibrate my d435i

AtariaLee, 2022-11-10:
**我按照流程使用`roslaunch d435i_xarm_setup d435i_xarm_auto_calib.launch`来对我的机器人和D435I相机进行手眼标定的时候，出现了错误。**
```
Planning scene monitors started.
[ INFO] [1668062504.006930064]: RealSense Node Is Up!
[ INFO] [1668062504.035976379]: Initializing OMPL interface using ROS parameters
[ INFO] [1668062504.050806547]: Using planning interface 'OMPL'
[ INFO] [1668062504.053747762]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1668062504.054227777]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1668062504.054458512]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1668062504.054903420]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1668062504.055144244]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1668062504.055379858]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1668062504.055411253]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1668062504.055423763]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1668062504.055437150]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1668062504.055454372]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1668062504.055471370]: Using planning request adapter 'Fix Start State Path Constraints'
[xarm_realsense_handeyecalibration_eye_on_hand/easy_handeye_calibration_server-12] process has died [pid 25423, exit code 1, cmd /home/lee/catkin_ws/src/easy_handeye/easy_handeye/scripts/calibrate.py __name:=easy_handeye_calibration_server __log:=/home/lee/.ros/log/bc867e5e-60c2-11ed-ae88-d83bbfe17db4/xarm_realsense_handeyecalibration_eye_on_hand-easy_handeye_calibration_server-12.log].
log file: /home/lee/.ros/log/bc867e5e-60c2-11ed-ae88-d83bbfe17db4/xarm_realsense_handeyecalibration_eye_on_hand-easy_handeye_calibration_server-12*.log
[ WARN] [1668062504.132577140]: 
[ INFO] [1668062504.268219815]: Stereo is NOT SUPPORTED
[ INFO] [1668062504.268273277]: OpenGL device: NVIDIA GeForce GTX 1660 Ti/PCIe/SSE2
[ INFO] [1668062504.268290611]: OpenGl version: 4.6 (GLSL 4.6).
arguments:  Namespace(quiet=False)
unknowns:  []
[INFO] [1668062504.372991]: Configuring for calibration with namespace: /xarm_realsense_handeyecalibration_eye_on_hand/
[INFO] [1668062504.374023]: Loading parameters for calibration /xarm_realsense_handeyecalibration_eye_on_hand/ from the parameters server
SDK_VERSION: 1.11.0
Tcp control connection successful
FIRMWARE_VERSION: v1.6.0, PROTOCOL: V1, DETAIL: h2,v1.6.0
Tcp Report Rich connection successful
Tcp Report Norm connection successful

```
**这个看起来是没有成功启动easy_handeye的服务，也没有进行标定的两个窗口。弹出的所有窗口是这个样子：**
[image]

penglongxiang, 2022-11-10:
您好，这不是xArm这边的问题，在开发视觉应用时涉及到很多第三方代码和库，需要您自己处理和解决，我们没有相应的视觉开发背景和经验。您可以按照错误提示定位到是哪个第三方代码的问题

vimior, 2022-11-11:
@AtariaLee 
```bash
 File "/home/lee/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibration_backend_opencv.py", line 4, in <module>
    import transforms3d as tfs
ImportError: No module named transforms3d
```
看这输出，是没有`transforms3d`这个库，可能是缺少某些依赖库

AtariaLee, 2022-11-11:
@vimior  这个问题我注意到了，并且已经解决，但是仍然会有这个报错，并不是这个的原因

closed 2022-11-24 by AtariaLee