xArm-Developer/xarm_ros #151: xArm gripper motion without moveit is not reflected in Rviz and "/joint_states" topic

HoneyMack, 2022-11-22:
Hello,
I've faced with a problem when trying to control real xArm7 gripper with Moveit and other api.
When I control a gripper with moveit, the gripper joint state is reflected to Rviz and /joint_states topic. However, when using other api, such as "rosservice call /xarm/gripper_move 500" and "UFACTORY studio", the visual state is not reflected in Rviz.

How can I reflect the joint state when using them?

ros nodes are bellow:
[image]

I want to refrect the openness of real robot motion.
[image]

Thank you.

vimior, 2022-11-22:
Hi, @HoneyMack 
This is indeed the case, because the mechanical arm and the gripper do not maintain communication all the time. At present, only when the gripper action of moveit is received in the driver can the gripper be controlled. During the movement of the gripper, the position of the mechanical claw will be continuously obtained and fed back to moveit. , the gripper will not continue to obtain the position of the gripper after it moves

HoneyMack, 2022-11-25:
@vimior sorry for the late reply. Thank you very much for your comment. 

And now, I'm trying to control two xArm7 with moveit and the two grippers attached to them with other API because I used the example "4. Dual xArm6 controlled with one moveGroup node" as a reference to use moveit for gripper control but I was not able to configure the gripper setting correctly.  So, I want to know the way to reflect the gripper joint state in order that two robots avoid collision when moving with moveit.
Do someone know how to reflect the gripper joint state?

Thank you.

vimior, 2022-12-13:
@HoneyMack 
Sorry for the late reply.
We did not provide topics related to the position of the gripper, but only configured GripperCommand to control the movement of the gripper when it receives an action. During the movement, it will feedback the position of the gripper until the gripper stops (the feedback stops), and the relevant feedback is topic It belongs to GripperCommand itself, you can check the instructions related to GripperCommand, or check the topic related to xarm_gripper/gripper_action.

HoneyMack, 2022-12-15:
@vimior 
Thank you very much for your advice.
I'll check the instructions related to GripperCommand and the topics related to xarm_gripper/gripper_action.