xArm-Developer/xarm_ros #156: Velocity mode VELO_DURATION as a parameter

BryanStuurman, 2023-01-07:
Currently VELO_DURATION in velocity mode is hardcoded to 1.0 seconds:

https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_controller/src/xarm_hw.cpp#L13

Is it possible to have this loaded in XArmHW::init() as a parameter, with a default value of 1.0 instead? I just tested a solution based on `robot_hw_nh.param<float>("velo_duration", velo_duration, 1.0);` and it works well.

I've been having some control computer hardware stability issues and subsequent robot collisions in my workspace, I'd like to have the robot stop a bit sooner when the computer crashes.

penglongxiang, 2023-01-09:
Hi @BryanStuurman, thanks for the suggestion and we will consider adding this new parameter in the next update. Right now, you may modifiy the source code to achieve the proper timeout for your application.

BryanStuurman, 2023-01-18:
Thanks for the fast turnaround with the patch.

Not sure if my comment on the patch has great visibility, so I'll reiterate here: I don't think the parameter server is robust in real time situations (while I can't find documentation prohibiting its use, I can't find documentation supporting it's realtime safety either).

I recommend getting the parameter once in init(). If  there's a desire to have this parameter reconfigurable while the robot is running a solution could be built off the dynamic_reconfigure package - though I confess I haven't used this myself.
http://wiki.ros.org/dynamic_reconfigure