xArm-Developer/xarm_ros #165: Lite 6 movement controlled by Moveit is not smooth

sean880304, 2023-02-13:
Recently I try to control Lite6 with Moveit by running planner and calling the service:

`$ roslaunch xarm_planner xarm_planner_rviz_sim.launch robot_dof:=<7|6|5> robot_type:=<xarm|lite>  add_gripper:=<true|false> add_vacuum_gripper:=<true|false>`
`$ rosservice call xarm_pose_plan 'target: [[0.28, 0.2, 0.2], [1.0, 0.0, 0.0, 0.0]]'`

But when I execute the trajectory, the movement of lite6 is not smooth. It seems to stop at some waypoints. The reason I may think of is it's probably caused somewhere during the transmission. I connect my laptop directly to Lite6 via Ethernet cable with an adapter transfers to usb (the adapter could be a reason?). And somehow there's some missing data or it transmits too slowly. Also I monitor the ping, its something like this:

```
64 bytes from 192.168.1.189: icmp_seq=2594 ttl=64 time=0.440 ms
64 bytes from 192.168.1.189: icmp_seq=2595 ttl=64 time=0.366 ms
64 bytes from 192.168.1.189: icmp_seq=2596 ttl=64 time=0.486 ms
64 bytes from 192.168.1.189: icmp_seq=2597 ttl=64 time=0.453 ms
64 bytes from 192.168.1.189: icmp_seq=2598 ttl=64 time=0.412 ms
64 bytes from 192.168.1.189: icmp_seq=2599 ttl=64 time=0.440 ms
64 bytes from 192.168.1.189: icmp_seq=2600 ttl=64 time=0.408 ms
64 bytes from 192.168.1.189: icmp_seq=2601 ttl=64 time=0.387 ms
64 bytes from 192.168.1.189: icmp_seq=2602 ttl=64 time=0.451 ms
64 bytes from 192.168.1.189: icmp_seq=2603 ttl=64 time=0.391 ms
64 bytes from 192.168.1.189: icmp_seq=2604 ttl=64 time=0.443 ms
64 bytes from 192.168.1.189: icmp_seq=2605 ttl=64 time=0.562 ms
64 bytes from 192.168.1.189: icmp_seq=2606 ttl=64 time=0.472 ms
64 bytes from 192.168.1.189: icmp_seq=2607 ttl=64 time=0.346 ms
64 bytes from 192.168.1.189: icmp_seq=2608 ttl=64 time=0.428 ms
64 bytes from 192.168.1.189: icmp_seq=2609 ttl=64 time=0.481 ms
64 bytes from 192.168.1.189: icmp_seq=2610 ttl=64 time=0.523 ms
64 bytes from 192.168.1.189: icmp_seq=2611 ttl=64 time=0.494 ms
```
Time is around 0.45ms, I don't know if it's normal?
Is there anyone experienced this similar issue before?
Thank you!

vimior, 2023-02-13:
@sean880304 
Hello, can you provide the command you are currently running, such as which launch file is started with what parameters, etc.
xarm_planner_rviz_sim.launch does not require a real machine, so it has nothing to do with adapters and networks.

sean880304, 2023-02-13:
oh sorry that was a typo, here are what i run:

`roslaunch xarm_planner xarm_planner_realHW.launch robot_ip:=192.168.1.189 robot_dof:=6 robot_type:=lite add_gripper:=true
`
`rosservice call xarm_pose_plan 'target: [[0.17, 0.02218, 0.07105], [1,0,0,0]]'
`
`rosservice call xarm_exec_plan 'true'
`
Thank you for helping me!

vimior, 2023-02-14:
@sean880304 
I can't reproduce this situation here, so it's not easy to evaluate. Can you take a video to show us the unsmooth situation, and send us the log by the way (including the log of ros and the log of the controller)
1. Synchronize the time through xArmStudio (Settings -> Timed Tasks -> Calibration) before testing, so that we can find the log of the controller
2. Record the current time
3. Start the test and record the video
4. Download the controller log through xArmStudio
5. Record the log of ros and give it to us together with the log of the controller, and inform the test time

sean880304, 2023-02-17:
@vimior 
Hi, sorry to ask, how do i get the ros log files?

vimior, 2023-02-18:
@sean880304 
You can directly copy the output of the terminal to a file.

sean880304, 2023-02-20:
@vimior 
Hi, thanks for the help!
I've already sent video and log files to you, please check.

sean880304, 2023-06-09:
[Problem update]

Still couldn't run the smooth trajectory by running rosservice command:
`$ rosservice call xarm_exec_plan 'true' `

But i managed to execute the planned trajectory smoothly by calling python API at every waypoints that generated from: 
`rosservice call xarm_pose_plan 'target: [[0.17, 0.02218, 0.07105], [1,0,0,0]]' `

And the python API is:
`arm.set_servo_angle(angle=angle, speed=10, mvacc=500, wait=False, radius=0.0)`

Just provide an alternative to those who may have similar problems.

closed 2023-06-09 by sean880304