xArm-Developer/xarm_ros #166: Lite Gripper Driver

pala5, 2023-02-16:
Hi,
I'm currently working with the Lite6 robot with the Lite Gripper coupled. I'm using ROS1 Noetic and Ubuntu 20.04 distributions.
I've faced with a problem when trying to control the Lite Gripper with the Moveit! configuration availabe in the  [xarm_ros](https://github.com/xArm-Developer/xarm_ros) github repo. 

There isn't a package for the Lite Gripper as it is for the [Xarm Gripper](https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_gripper) one, with the necessary transmission and URDF configurations files.

I've checked that for the Lite Gripper it is only available a [stl file](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_description/meshes/lite6/visual/gripper_lite.stl) that includes the whole gripper hardware, without distinguish between the fixed and the actuated parts, as it is done for the [Xarm Gripper](https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_gripper/meshes).

At this point, is not possible to move the gripper within the MoveIt! functionality. There is available any package where I can find the necessary configuration files for using the Lite Gripper with ROS and MoveIt!

Thanks.

vimior, 2023-02-17:
@pala5 
Yes, currently only the gripper of xarm supports moveit control in the ROS package, and the gripper of Lite6 has not yet supported it.

pala5, 2023-02-17:
@vimior 
Are you planning to implement the Lite6 Moveit control package in the near future? If not, could you suggest an available alternative?

Thanks!

vimior, 2023-02-24:
@pala5 
I'm sorry, I'm currently busy with my work schedule, so I don't have this plan for now. At present, Lite Gripper cannot be used with moveit. It can only be controlled through service (not implemented yet) or SDK. If there is a need to control through service, we can provide the implementation of such service.

pala5, 2023-02-24:
@vimior 
If it is not an arduous task for you, it would help me a lot. 
Thansk!

samhita-m, 2023-06-26:
I tried making a Moveit package for the lite6 arm with its gripper by separating out the fingers with the gripper_base but I am stuck at initializing the gripper controller as shown below. 

```
[ERROR] [1687756001.280765683, 0.468000000]: Could not find joint 'drive_joint' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1687756001.280816080, 0.468000000]: Failed to initialize the controller
[ERROR] [1687756001.280833657, 0.468000000]: Initializing controller 'gripper_trajectory_controller' failed
```
I followed a similar format as other ()_gripper_moveit_config but it looks like I need a controller specific for lite_gripper. Can I get some direction in figuring this out? Thanks!

vimior, 2023-06-26:
@samhita-m 
Sorry, the structure of Lite Gripper does not support this control method. Currently, it cannot be controlled through moveit. Currently, the opening and closing of Lite Gripper can only be controlled through services.

samhita-m, 2023-06-26:
Got it, will I be still be able to see the grippers opening and closing in Gazebo if I use services? 

As an alternative, can I connect xArm gripper with Lite6 arm on the real system?

vimior, 2023-06-26:
@samhita-m 
Gazebo can't either, the service currently provided can only control the real machine.
The interfaces of Lite Gripper and xArm Gripper are different, so it is impossible to connect xArm Gripper to Lite6 for use.

samhita-m, 2023-06-27:
Thanks for confirming @vimior!

I'm not quite familiar with ros controllers but looking through the tutorials like [here](https://www.theconstructsim.com/ros-qa-070-moving-joints-gazebo-simple-example/), it seems like I need to declare `hardwareInterface` through `transmission` and have the plugin `libgazebo_ros_control.so`. I'm confused as to why I can't open and close the gripper by following these instructions. Could you elaborate on what additional files/drivers are required?

pala5, 2023-06-27:
@samhita-m 
Hi, correct my @vimior if I am wrong. The ROS services only work with the real robot because they are using the Xarm_SDK_API, not the HardwareInterface from ROS Control. 
I do not have time for doing so, but look at the xArm Gripper packages, maybe there they are going to be its implementation. The lite6 one should be similar.
Mainly, the files that you need to do that are: 
- HardwareInterface
- Transmissions
Then, in the URDF of the gripper, you will need to declare the actuated joints like revolute, not fixed.
And finally the general gazebo add-ons that you mentioned.

samhita-m, 2023-06-28:
Thanks for the guidelines @pala5! I seem to make the original error go away (Gazebo is able to find the gripper_controller) but now I'm stuck at the following error when I run `roslaunch lite6_gripper_moveit_config lite6_gripper_moveit_gazebo.launch add_gripper:=true
.` 

[image]

The gripper fingers close and open in Rviz when I click plan but because of the error above, it is not being executed. Any idea why this might be happening?

[image]

vimior, 2023-06-29:
@samhita-m 
As @pala5 said, the structure of the Lite Gripper is different from that of the xArm Gripper. The driving joints of the Lite Gripper are not defined in the URDF, so the model you see is just a whole and cannot be controlled, even if it is the opening and closing of the real mechanical claw. Controls, there won't be any changes to this model either.

vatozZ, 2025-05-24:
Are there any new recommendations regarding a gripper for the Lite6? Has there been any update regarding MoveIt support?