xArm-Developer/xarm_ros #168: Triggering the force torque sensor gravity compensation calibration routine from ROS

VFjr, 2023-02-17:
Hi, we would greatly appreciate if it would be possible to trigger the force torque sensor gravity compensation calibration routine from ROS, rather than having to use Ufactory Studio to do it manually.

Do you think that this could be a potential future feature in the ROS driver?

vimior, 2023-02-18:
@VFjr 
This is possible, currently only the following torque interface services are provided in xarm_ros: `ft_sensor_enable`, `ft_sensor_app_set`, `ft_sensor_set_zero`, `ft_sensor_cali_load`, `get_ft_sensor_error`
If necessary, you can add some torque-related services in the future. You can refer to the [SDK documentation](https://github.com/xArm-Developer/xArm-CPLUS-SDK/blob/master/doc/xarm_cplus_api.md) to see which torque interfaces need to be used on ROS. Let’s evaluate

VFjr, 2023-02-20:
Thank you. 

Just to confirm, now the `ft_sensor_iden_load` service loads the parameters that are saved in the control box, right?

Therefore we would only need to run the force torque sensor calibration routine in uFactory Studio calibration just once, and afterwards we would be able to use `ft_sensor_iden_load` to fetch the parameters every time when starting the ROS driver?

vimior, 2023-02-21:
@VFjr 
`ft_sensor_iden_load` is used for identification, and the result of identification can be made effective through `ft_sensor_cali_load` (adding the operation of writing to the configuration file).
The combined operation is similar to that of the studio.

VFjr, 2023-02-28:
@vimior  

I tried using `ft_sensor_iden_load`, and it did go through the full calibration procedure with the arm moving in a similar way as to when doing the identification through xArm Studio. 

Unfortunately when the calibration is done, I am getting an array full of 0s as the result.

```
$ rosservice call /xarm/ft_sensor_iden_load "{}" 
ret: 0
message: "ft_sensor_iden_load, ret = 0"
datas: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

vimior, 2023-03-01:
@VFjr 
Whether the torque sensor is not enabled (`ft_sensor_enable`), it needs to be enabled first.

nitish3693, 2023-03-22:
@vimior Thanks for the help. 

With the force torque sensor enabled we were able to get the calibration values using the service `ft_sensor_iden_load`
```
rosservice call /xarm/ft_sensor_iden_load
ret: 0
message: "ft_sensor_iden_load, ret = 0"
datas: [1.0952531099319458, -0.24829700589179993, -1.9048782587051392, 63.123634338378906, -7.2495903968811035, 1.3315763473510742, 20.2039852142334, -1.5803824663162231, -0.11448634415864944, 0.3574244976043701]

```