xArm-Developer/xarm_ros #172: Hand-eye Calibration for LITE 6

sean880304, 2023-02-24:
Hi, I currently run the calibration launch file for lite6:

`$ roslaunch d435i_xarm_setup d435i_xarm_auto_calib.launch robot_dof:=6 robot_ip:=192.168.1.xxx`

But the model appears in the RViz seems like xarm6 instead of lite6. Therefore, the plan is not suitable for lite6. Where should I modify to change to lite6 model?

Thanks!!

sean880304, 2023-02-24:
@Akumar201 Thanks I'll try this out.

penglongxiang, 2023-02-28:
hi @sean880304 , we have added the corresponding launch file for lite6 calibration with realsense d435i, please check [here](https://github.com/xArm-Developer/xarm_ros#721-hand-eye-calibration-demo-for-ufactory-lite6).

sean880304, 2023-03-02:
@penglongxiang Thank you, it works now.

closed 2023-03-02 by sean880304

Murali25111993, 2024-07-29:
Hi @Akumar201 ,

I tried with eye_on_hand, working find but I could not able to provide proper quaternion for this below setup. If possible please provide an idea

[image]