xArm-Developer/xarm_ros #177: Large Detours on Trajectories Found by MoveIt Planner

febert, 2023-03-14:
Hi,

I'm trying to use xarm_planner to plan around a couple of obstacles.
However the planner finds paths that are highly suboptimal and take very large detours, as shown in the picture below:

[image]

Are there any planner settings or alternative planners that I can try? It should be using RRTConnect by default, is that correct?

Repeatedly sampling plans does yield shorter paths that would be satisfactory for my application. But with the movegroup interface it is quite slow to run in a loop since `group.plan(my_xarm_plan)` performs visualization for every single plan.

Thanks a lot!

penglongxiang, 2023-03-16:
Hi, xarm_planner is a wrapper for the standard movegroup interface, you may check the config file or the parameters loaded and change the planner configuration. Are you using xarm7 for simulation?

febert, 2023-03-21:
Hi Penglongxiang,

I'm using the following command `roslaunch xarm_planner xarm_planner_rviz_sim.launch robot_dof:=7 robot_type:=xarm` as described in the docs.

In which file can I choose the type of planner and change the parameters?

I found this file: https://github.com/xArm-Developer/xarm_ros/blob/master/xarm7_moveit_config/config/ompl_planning.yaml
but I don't see where to change the default solver.

Thanks!

febert, 2023-03-21:
Also, is there a plan to create a MoveIt2 support for xarm7? It seems to readily support running multiple different planners in parallel and you can specify the number of planning attempts, which should directly solve my problem.

penglongxiang, 2023-03-21:
Hi, you may add "default_planner_config" parameter like this file: https://github.com/xArm-Developer/xarm_ros/blob/master/xarm6_moveit_config/config/ompl_planning.yaml. We support Moveit2 in our ROS2 repository, however we did not test or try using Moveit2 with ROS1.