xArm-Developer/xarm_ros #181: xarm6_with_gripper  srdf

chojo1, 2023-03-27:
I want to modify the configuration of moveit_setup_assistant for xarm6_with_gripper
But I found it missing xarm6_with_gripper srdf，Is there anyone who can share the srdf file with me
really appreciate

penglongxiang, 2023-03-29:
Hi, currently many files in our moveit_config packages are not auto-generated, so the original .setup_assistant configuration files are no longer valid. If you would like to change the configuration, you may have to do it manually, now all the URDF and SRDF files used are located at xarm_description/urdf and xarm_description/srdf.

If you have any difficulties, please tell us what configuration needs to be changed and we will assist you accordingly. Thanks.

chojo1, 2023-03-29:
@penglongxiang Thanks for your reply, I will try to modify the configuration file manually