xArm-Developer/xarm_ros #183: xarm6 Hand-eye Calibration Demo Result

KylecheungKml, 2023-04-14:
We are attempting to do the "7.2 Hand-eye Calibration Demo" using a 3D-print UFACTORY camera stand and compare our output with the sample calibration results provided by you. However, we have observed a difference between our output and the sample calibration results. The hand-eye calibration results seem to translate the camera to the centre location of the end effector at different z positions but not the camera position in the real world. The z-axis distance offset is correct.  Can you please provide assistance in resolving this issue? Thanks.

[image]
[image]
[image]

penglongxiang, 2023-04-14:
Hi, it seems your calibration result is off especially in the Z direction. Even our sample result is not 100% accurate, it still serves as a reference and the correct/usable result should not deviate too much from it. Please re-check your whole calibration process and consult "easy_handeye" developer team if necessary.

The camera_color_optical_frame shown in your second picture during calibration process is not from the final result, but a dummy fixed position (only a non-zero z offset) since it is not known before calibration is performed.  The visualization would not be correct unless you publish the calibrated TF relationship (from end_effector to camera) afterwards in your vision application.

penglongxiang, 2023-04-14:
For the real world position of “camera_color_optical_frame”, please refer to this reply:  https://github.com/xArm-Developer/xarm_ros/issues/132#issuecomment-1272433065

KylecheungKml, 2023-04-18:
Thank you for your help. After modifying the static_transform_publisher of dummy_handeye in the calibrate.launch file to 0 0 0.05 1.5708 0 0, I was able to achieve a result similar to the sample result shown in RVIZ.
[image]

However, when I ran the command "ros2 launch xarm_planner xarm6_planner_realmove.launch.py robot_ip:=192.168.1.237 add_realsense_d435i:=true" in the terminal, I noticed that the camera_color_optical_frame was different from the result obtained in ROS1 and some camera frame is out of the camera model. I am unsure if this is an issue.
[image]

KylecheungKml, 2023-04-18:
Thank you for your help. After modifying the static_transform_publisher of dummy_handeye in the calibrate.launch file to 0 0 0.05 1.5708 0 0, I was able to achieve a result similar to the sample result shown in RVIZ.
[image]

However, when I ran the command "ros2 launch xarm_planner xarm6_planner_realmove.launch.py add_realsense_d435i:=true" in the terminal, I noticed that the camera_color_optical_frame was different from the result obtained in ROS1 and some camera frames are out of the camera model. I am unsure if this is an issue.
[image]

closed 2023-04-18 by KylecheungKml