xArm-Developer/xarm_ros #191: "run_demo" launch argument doesn't exist

jkaniuka, 2023-07-13:
In README section 4.7 (_Run the demo in Gazebo simulator_) there is the following command:
```
roslaunch xarm_gazebo xarm7_beside_table.launch [run_demo:=true] [add_gripper:=true] [add_vacuum_gripper:=true] 
```
Unfortunately, in the `xarm_gazebo/launch/xarm7_beside_table.launch` file, the **run_demo** argument is not declared.

vimior, 2023-07-13:
@jkaniuka 
Thank you for reminding, because a new launch file was referenced, this parameter was removed and it was unavailable. Now the support for this parameter has been restored, please use it after pulling the update.

jkaniuka, 2023-07-13:
Thanks, robot now executes sample trajectory in the simulation as expected :+1:

closed 2023-07-13 by jkaniuka