xArm-Developer/xarm_ros #195: rviz無法顯示機械手臂

frinklai, 2023-08-03:
您好，我參考readme[教程](xarm7_rviz_display)內容運行
`roslaunch xarm7_moveit_config demo.launch`後，發現沒有顯示機械手臂在rviz中

如果透過
` roslaunch xarm_gazebo xarm7_beside_table.launch`
以及
`roslaunch xarm7_moveit_config xarm7_moveit_gazebo.launch`

會開啟rviz與gazebo，其中，gazebo中會有手臂，但rviz中則不會
請問是否有方法可以在運行launch後，顯示手臂在rviz上
感謝回覆

penglongxiang, 2023-08-04:
您好，正常情况下是可以显示的。如图：
[image]

是否有部分组件没有完整安装？可否检查一下launch之后terminal里面的详细输出，如果有任何警告或者错误信息，请反馈给我们，谢谢。

frinklai, 2023-08-05:
您好，我運行`˙roslaunch xarm7_moveit_config demo.launch`後，看起來是沒有特別的錯誤
有輸出**No root/virtual joint specified in SRDF. Assuming fixed joint** 的訊息，但他是info層級
另外補充，我有發現source code ROS版本是使用kinetic，但我使用的是 noetic，這部分是否有影響呢

完整輸出如下
```
You can start planning now!

[ INFO] [1691228688.496985180]: Stereo is NOT SUPPORTED
[ INFO] [1691228688.497057888]: OpenGL device: D3D12 (NVIDIA GeForce RTX 3080)
[ INFO] [1691228688.497089617]: OpenGl version: 3.1 (GLSL 1.4).

[ INFO] [1691228691.849759994]: Loading robot model 'UF_ROBOT'...  
[ INFO] [1691228691.849818780]: No root/virtual joint specified in SRDF. Assuming fixed joint <---是否這裡有問題

[ INFO] [1691228692.015204432]: Starting planning scene monitor
[ INFO] [1691228692.016991641]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1691228692.029333171]: Constructing new MoveGroup connection for group 'xarm7' in namespace ''
[ INFO] [1691228693.030013510]: Ready to take commands for planning group xarm7.
```

此外，我有運行`roslaunch moveit_setup_assistant setup_assistant.launch`
載入`xarm7_gripper_moveit_config`，發現UI右側沒有顯示手臂(如圖)

[image]

[image]

penglongxiang, 2023-08-07:
您好，不知道是否与使用的NVIDIA GPU有关，可以尝试关闭GPU来看看能不能正常显示。

penglongxiang, 2023-08-07:
我这边不了解您的运行系统环境，有WSLg (Windows Subsystem for Linux GUI) 的用户反馈过类似的问题， 参考: https://github.com/microsoft/wslg/issues/554