xArm-Developer/xarm_ros #196: Unable to conver xarm7.urdf.xacro to URDF file

vivekcdavid, 2023-08-09:
I want to convert the xarm7.urdf.xacro file to a pure URDF file which I need for creating the IKfast plugin for the xarm7 robot. But the method to conver from .xacro.urdf to urdf is giving just the lines outside of the
 <xacro:macro .....> 
.....
...

rosrun xacro xacro -o /tmp/$MYROBOT_NAME.urdf ~/xarm_ws/src/xarm_ros/xarm_description/urdf/xarm7/xarm7.urdf.xacro

vimior, 2023-08-10:
@vivekcdavid 
The wrong file was transferred. There are only macros defined in xarm7.urdf.xacro, but no calls. You should transfer (latest version) `xarm_description/urdf/xarm_device.urdf.xacro` or (previous version) `xarm_description/urdf/xarm7_robot.urdf .xacro`

vivekcdavid, 2023-08-11:
Thank you so much, that worked.

closed 2023-08-11 by vivekcdavid