xArm-Developer/xarm_ros #198: Xarm simulation in gazebo not working

HoteSaing, 2023-08-10:
Hi, I have been trying to run gazebo simulation on xarm according to the official guide here: https://github.com/xArm-Developer/xarm_ros. However, after running the simulation for a few seconds the manipulator just falls to the ground and I cannot do anything about it. It would be much appreciated if you could help me. Here is my error message:

$ roslaunch xarm_gazebo xarm7_beside_table.launch
... logging to /home/lyb/.ros/log/e59d946c-3777-11ee-9446-f147a27eb3fa/roslaunch-lyb-2666363.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
started roslaunch server http://lyb:36797/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /use_sim_time: True
 * /xarm/DOF: 7
 * /xarm/joint_names: ['joint1', 'joint...
 * /xarm/joint_state_controller/publish_rate: 50
 * /xarm/joint_state_controller/type: joint_state_contr...
 * /xarm/xarm7_traj_controller/action_monitor_rate: 10
 * /xarm/xarm7_traj_controller/constraints/goal_time: 0.5
 * /xarm/xarm7_traj_controller/constraints/joint1/goal: 0.01
 * /xarm/xarm7_traj_controller/constraints/joint1/trajectory: 1
 * /xarm/xarm7_traj_controller/constraints/joint2/goal: 0.01
 * /xarm/xarm7_traj_controller/constraints/joint2/trajectory: 1
 * /xarm/xarm7_traj_controller/constraints/joint3/goal: 0.01
 * /xarm/xarm7_traj_controller/constraints/joint3/trajectory: 1
 * /xarm/xarm7_traj_controller/constraints/joint4/goal: 0.01
 * /xarm/xarm7_traj_controller/constraints/joint4/trajectory: 1
 * /xarm/xarm7_traj_controller/constraints/joint5/goal: 0.01
 * /xarm/xarm7_traj_controller/constraints/joint5/trajectory: 1
 * /xarm/xarm7_traj_controller/constraints/joint6/goal: 0.01
 * /xarm/xarm7_traj_controller/constraints/joint6/trajectory: 1
 * /xarm/xarm7_traj_controller/constraints/joint7/goal: 0.01
 * /xarm/xarm7_traj_controller/constraints/joint7/trajectory: 1
 * /xarm/xarm7_traj_controller/constraints/stopped_velocity_tolerance: 0.05
 * /xarm/xarm7_traj_controller/joints: ['joint1', 'joint...
 * /xarm/xarm7_traj_controller/state_publish_rate: 25
 * /xarm/xarm7_traj_controller/stop_trajectory_duration: 0.2
 * /xarm/xarm7_traj_controller/type: position_controll...
 * /xarm/xarm7_traj_controller_velocity/action_monitor_rate: 10
 * /xarm/xarm7_traj_controller_velocity/constraints/goal_time: 0.5
 * /xarm/xarm7_traj_controller_velocity/constraints/joint1/goal: 0.01
 * /xarm/xarm7_traj_controller_velocity/constraints/joint1/trajectory: 1
 * /xarm/xarm7_traj_controller_velocity/constraints/joint2/goal: 0.01
 * /xarm/xarm7_traj_controller_velocity/constraints/joint2/trajectory: 1
 * /xarm/xarm7_traj_controller_velocity/constraints/joint3/goal: 0.01
 * /xarm/xarm7_traj_controller_velocity/constraints/joint3/trajectory: 1
 * /xarm/xarm7_traj_controller_velocity/constraints/joint4/goal: 0.01
 * /xarm/xarm7_traj_controller_velocity/constraints/joint4/trajectory: 1
 * /xarm/xarm7_traj_controller_velocity/constraints/joint5/goal: 0.01
 * /xarm/xarm7_traj_controller_velocity/constraints/joint5/trajectory: 1
 * /xarm/xarm7_traj_controller_velocity/constraints/joint6/goal: 0.01
 * /xarm/xarm7_traj_controller_velocity/constraints/joint6/trajectory: 1
 * /xarm/xarm7_traj_controller_velocity/constraints/joint7/goal: 0.01
 * /xarm/xarm7_traj_controller_velocity/constraints/joint7/trajectory: 1
 * /xarm/xarm7_traj_controller_velocity/constraints/stopped_velocity_tolerance: 0.01
 * /xarm/xarm7_traj_controller_velocity/gains/joint1/d: 0.0
 * /xarm/xarm7_traj_controller_velocity/gains/joint1/i: 0.0
 * /xarm/xarm7_traj_controller_velocity/gains/joint1/i_clamp: 1
 * /xarm/xarm7_traj_controller_velocity/gains/joint1/p: 10
 * /xarm/xarm7_traj_controller_velocity/gains/joint2/d: 0.0
 * /xarm/xarm7_traj_controller_velocity/gains/joint2/i: 0.0
 * /xarm/xarm7_traj_controller_velocity/gains/joint2/i_clamp: 1
 * /xarm/xarm7_traj_controller_velocity/gains/joint2/p: 10
[... 66 lines cut ...]
[ INFO] [1691669860.228283340, 0.173000000]: Loading gazebo_ros_control plugin
[ INFO] [1691669860.228402253, 0.173000000]: Starting gazebo_ros_control plugin in namespace: /xarm
[ INFO] [1691669860.228734254, 0.173000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ERROR] [1691669860.355144719, 0.173000000]: Did not find PID configuration for joint 'joint1'.
Commands from 'hardware_interface::PositionJointInterface' will bypass dynamics.
[ERROR] [1691669860.355426097, 0.173000000]: Did not find PID configuration for joint 'joint2'.
Commands from 'hardware_interface::PositionJointInterface' will bypass dynamics.
[ERROR] [1691669860.355538144, 0.173000000]: Did not find PID configuration for joint 'joint3'.
Commands from 'hardware_interface::PositionJointInterface' will bypass dynamics.
[ERROR] [1691669860.355631218, 0.173000000]: Did not find PID configuration for joint 'joint4'.
Commands from 'hardware_interface::PositionJointInterface' will bypass dynamics.
[ERROR] [1691669860.355718136, 0.173000000]: Did not find PID configuration for joint 'joint5'.
Commands from 'hardware_interface::PositionJointInterface' will bypass dynamics.
[ERROR] [1691669860.355803253, 0.173000000]: Did not find PID configuration for joint 'joint6'.
Commands from 'hardware_interface::PositionJointInterface' will bypass dynamics.
[ERROR] [1691669860.356020510, 0.173000000]: Did not find PID configuration for joint 'joint7'.
Commands from 'hardware_interface::PositionJointInterface' will bypass dynamics.
[ INFO] [1691669860.356235199, 0.173000000]: Joint mode switching is enabled
[ INFO] [1691669860.361156587, 0.173000000]: Loaded gazebo_ros_control.

[spawn_gazebo_model-4] process has finished cleanly
log file: /home/lyb/.ros/log/e59d946c-3777-11ee-9446-f147a27eb3fa/spawn_gazebo_model-4*.log
../src/intel/isl/isl.c:2105: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.

[image]

penglongxiang, 2023-08-11:
Hi, according to the output:
```
/usr/bin/env: ‘python’: No such file or directory
process[robot_state_publisher-6]: started with pid [2666525]
[xarm/traj_controller_spawner-5] process has died [pid 2666524, exit code 127, cmd /home/lyb/catkin_ws/src/ros_control/controller_manager/scripts/controller_manager spawn xarm7_traj_controller joint_state_controller __name:=traj_controller_spawner __log:=/home/lyb/.ros/log/e59d946c-3777-11ee-9446-f147a27eb3fa/xarm-traj_controller_spawner-5.log].
```
Have you properly installed python? In order to make sure the proper execution, please also make sure you have installed everything mentioned in readme Chapter 3.

HoteSaing, 2023-08-11:
Hi, I have solved my issue. I found out that there are ros-control and ros-controllers packages installed in my another workspace (catkin_ws) and it is somehow effecting my program in this workspace (xarm_ws). Seems like the program, by default, checks for the packages in the catkin_ws first and only after that it checks the package in /opt/ros directory. I removed those packages from catkin_ws and the problem is solved.

closed 2023-08-11 by HoteSaing