xArm-Developer/xarm_ros #200: xarm planner simulation

Tn-333, 2023-09-05:
I'm a beginner, so I apologize for the basic questions.

`roslaunch xarm_planner xarm_planner_rviz_sim.launch robot_dof:=7 robot_type:=xarm add_gripper:=false add_vacuum_gripper:=false`

When I start rviz with this command and run `xarm_simple_planner_test`

> [ INFO] [1693880039.867058739]: waitForService: Service [/xarm/set_state] has not been advertised, waiting...

It will be displayed. Is this normal or abnormal?
If it is abnormal, please tell me the solution.

vimior, 2023-09-05:
@Tn-333 
Sorry, this `xarm_simple_planner_test` routine only supports real robotic arms. Virtual robotic arms can directly call the corresponding services through `rosservice call`. In fact, `xarm_simple_planner_test` also calls some services, but there is a service that only supports real robots. The virtual manipulator only has the arm, so when the virtual manipulator runs this routine, it will be stuck in the connection service.

Tn-333, 2023-09-11:
Thank you for your detailed reply.
I would like to ask you something additional.
I would like to run C++ code and check its operation without connecting to an actual xarm7 machine. What steps should I take?

vimior, 2023-09-11:
@Tn-333 
As you used before, you started `xarm_planner_rviz_sim.launch`. This will not be connected to the real robot arm. However, `xarm_simple_planner_test` does not apply to `xarm_planner_rviz_sim.launch`. You can simply modify the source code of xarm_simple_planner_test and compile it for use, or call it directly through the command line. services control, refer to [xarm_planner](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_planner/ReadMe.md).