xArm-Developer/xarm_ros #206: UFactory Lite 5 Model

DasGuna, 2024-01-12:
Are the URDF files for the lite 5 available (or similar to that of the xarm 5)? Thanks in advance.

penglongxiang, 2024-01-12:
Hi, the `.xacro` files are available under `xarm_ros/xarm_description/urdf/xarm5` directory, if pure `.urdf` file is needed, please check the attached file(please modify the name to 'xarm5.urdf' or it will fail to upload here), it can be automatically generated by navigating to `xarm_ros/xarm_description/urdf` directory and execute: 

`rosrun xacro xacro xarm_device.urdf.xacro robot_type:=xarm dof:=5 --inorder > xarm5.urdf`

[xarm5.urdf.txt](https://github.com/xArm-Developer/xarm_ros/files/13915346/xarm5.urdf.txt)

DasGuna, 2024-01-12:
Thanks for the clarification!

closed 2024-01-12 by DasGuna