xArm-Developer/xarm_ros #207: Robot Mode 7: Velocity Limit & Null-Space Optimization

lenzc, 2024-01-15:
Hi,
I'm experimenting with controlling an xArm7 in mode 7 by calling [`MoveAxisAngle`](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_msgs/srv/MoveAxisAngle.srv) services and got two questions:
1. How is a single parameter, `mvvelo`, used to limit the motion velocity in both translation and rotation? Does the controller have an internal conversion between `mm/sec` and `rad/sec`? (The same applies to the acceleration limit).
2. In the case of the xArm7, does the internal controller perform any kind of null-space optimization?

penglongxiang, 2024-01-16:
Hi @lenzc, 

1. Sorry just realized our documentation in the service file is not very appropriate. Please refer to the notice in our developer manual:
[image]
In short, the range for velocity parameter is **0 to 1000** (mapped from smallest to largest allowed) **for both linear and angular speed**. 
For acceleration, the controller uses a fixed value for angular and the user specified value (mm/s^2) for linear.

2. There is currently no direct support for null-space operation for xArm7 by our service, we will evaluate and see if this function can be added in future releases. There is one related example (in simulation) contributed by other users, you can check [here](https://github.com/xArm-Developer/xarm_ros/tree/master/examples/xarm7_redundancy_res).

lenzc, 2024-01-16:
Thanks @penglongxiang for the fast reply. This answers my questions.

closed 2024-01-16 by lenzc