xArm-Developer/xarm_ros #208: xarm_planner goal flashes from time to time and service not available

lemonci, 2024-02-01:
System: Ubuntu 20.04+ROS noetic
Commands executed:
`roslaunch xarm_gazebo xarm5_beside_table.launch`
Gazebo works fine
`roslaunch xarm_planner xarm_planner_rviz_sim.launch robot_dof:=5 robot_type:=xarm  add_gripper:=false add_vacuum_gripper:=false`
Rviz starts successfully, but when I add the MotionPlanning panel and select the Planning Group as xarm5, the goal position flashes on the field, like the following screenshots show:
[image]
[image]
Also, the services under xarm_planners are not available:
[image]
When I worked on the real robots, the service under mode 0 works though.

penglongxiang, 2024-02-02:
Hi, please check the instruction [here](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#to-run-moveit-motion-planner-along-with-gazebo-simulator), if you would like to use motion planner to control the model in Gazebo, you should run` $ roslaunch xarm5_moveit_config xarm5_moveit_gazebo.launch` instead of the launch file in `xarm_planner` package, which is not supposed to be connected with Gazebo.

lemonci, 2024-02-02:
Hello, I just wonder if it is possible to use `xarm_planner` without a real robot.
First I launched `roslaunch xarm_planner xarm_planner_rviz_sim.launch robot_dof:=5 robot_type:=xarm  add_gripper:=false add_vacuum_gripper:=false` and then I called the service `rosservice call xarm_joint_plan 'target: [1.0, -0.5, 0.0, -0.3, 0.0, 0.0, 0.5]'` and it returns `success:True` but I see nothing in Rviz and the goal control in Rviz keeps flashing. Thanks.

penglongxiang, 2024-02-03:
The cause may be the dimension of joint targets in your command, please **change it to 5 numbers instead of 7**. It turns out the 
plan() function in moveit interface will still return success even if the given dimension is not correct, however, the actual trajectory is not valid.

penglongxiang, 2024-02-03:
you may check out the latest commit, I added the logic to prevent joint dimension mismatch.

lemonci, 2024-02-06:
Hello, I have pull the repo and compiled the source code. However, the goal position flashing problem still appears in Rviz no matter if I use dof=5 or dof=7. Also, since our arm is xArm5, it does not make sense to use an xArm7 model for simulation.

Thanks.

penglongxiang, 2024-02-18:
Please provide full information of your launching command this time in order, or the backstage abnormal output if any. Otherwise we could not reproduce the issue. Also, please follow the instructions in ReadMe strictly for different scenarios. Some of the launch files should not be mixed or used together.  For example, we did not provide gazebo integration demo for "xarm_planner" package, it can be simulated in **Rviz** only, or used to control the real xArm. If gazebo and xarm_planner are still both launched like you initially described, abnormal things like name conflicts may occur and the commands may not work as expected. If gazebo + xarm_planner is a must for your project, you may develop the integration on your own.