xArm-Developer/xarm_ros #220: control pose/velocity vs measured pose/velocity and tf

bmegli, 2024-03-17:
From:
- https://github.com/xArm-Developer/xarm_ros/issues/25#issuecomment-1614378290
  - and later/earlier posts

it looks like the data available through ROS is control pose/velocity, not the measured pose/velocity.

1. Is it still the case? (looks it was at end of 2023)
2. Do I understand correctly that joint_states are filled from control data (vs measured) and in consequence so is ROS tf data?
4. If above, what difference should be expected between control/measured pose/velocity from user side? 
5. If above, are there any plans to share measured pose/velocity over ROS?

Kind regards!

penglongxiang, 2024-03-18:
Hi, to better assist with your issue, may I ask what is the controlling method you plan to use in ROS? Is it through **Moveit** or our wrapped ROS **services**?

bmegli, 2024-03-18:
Hi @penglongxiang,

MoveIt like, sending controller commands to the trajectory controller topic.

But I am specifically interested in
- what exactly is published over ROS `/tf` topic
  - which likely has data from `/joint_states`

So the question is about time/pose synchronization using `/tf`with other subsystem rather then control itself.

Will the control vs measured pose/velocity on ROS side affect us in such scenario etc?

penglongxiang, 2024-03-19:
Hi @bmegli, so you are using the `ros-controller` and the `hardware interface` to our robot. We have just optimized for this scenario and the `/joint_states` published will be using measured joint angles, so as the joint velocity. It will be available in the **next controller firmware version 2.4.0** which is supposed to be released by the end of this month. `/tf` result is derived from `/joint_states` and urdf model, so the source of `/tf` will also be updated. 

BTW if you need less delay or higher update rate, please specify `report_type:=dev` at launching stage, the publish rate will increase from 5Hz to 100Hz. Please also update the 'xarm_ros' repository to the latest since we will keep optimizing and adding new functions.

bmegli, 2024-03-19:
Thank you @penglongxiang 

From XARM side we are using [realMove_exec.launch](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm6_moveit_config/launch/realMove_exec.launch) with `xarm6_traj_controller`

> BTW if you need less delay or higher update rate, please specify report_type:=dev

Yes, we are already doing it as well as bumping `jnt_stat_pub_rate` to get `/tf` data at higher frequency

>  We have just optimized for this scenario and the /joint_states published will be using measured joint angles, so as the joint velocity. It will be available in the next controller firmware version 2.4.0 which is supposed to be released by the end of this month

Ok! In that case I will keep the issue open until new firmware is released

penglongxiang, 2024-04-03:
Hi @bmegli, I am sorry to say that our official firmware release will be delayed. If you are eager to test the measured joint angles as feedback, I can send you a beta version that already have this feature early next week.

bmegli, 2024-04-24:
Thank you @penglongxiang 

I think we will wait for the stable version

bmegli, 2024-06-24:
> Has the Hi @bmegli, I am sorry to say that our official firmware release will be delayed. If you are eager to test the measured joint angles as feedback, I can send you a beta version that already have this feature early next week.

Has the new firmware with measured pose/velocity/tf been eventually released?

penglongxiang, 2024-06-25:
Yes, it has been released already, sorry I forgot to notify you in particular. Please follow the update instruction inside the user manual (Appendix 4) and have a test.

nitish3693, 2024-07-03:
@penglongxiang  Do use this feature do we need to use the latest ROS driver from the repository,  or does it work with old versions (old commits) from the repository.

penglongxiang, 2024-07-04:
@nitish3693  It depends on how old your local commit is, it is recommended to use the latest or recent commits for better compatibility with the new firmware.