xArm-Developer/xarm_ros #221: Xarm7 Robot deviating from the target pose - problem with URDF? Or Encoder error?

vivekcdavid, 2024-03-28:
When trying to change the orientation of the end effector while keeping the same cartesian pose the real robot moves in its Y-axis of the end effector frame by about 1cm for a 90-degree rotation around the Y-axis of the end-effector. I tried calibrating the TCP initially, but the issue persisted, so that is not the issue. I've done many tests and the issue persists even without the end effector attached and running the default version of the realMove_exec.launch.

Has this behavior been reported before?  I suspect that it is an issue either in the default URDF provided in the xarm_ros package or it is an issue of the encoders of the hardware. But I don't know how to diagnose the issue further. 

or is there any way to test the calibration for each of the joints and its encoders?

Any help in identifying the issue is really helpful.

Thanks in advance.

penglongxiang, 2024-03-29:
Hi @vivekcdavid, this is mainly due to the error of real mechanical parts and minor inconsistency introduced by assembly. The actual kinematics will differ slightly from the theoretical model defined in URDF. 

May I have the **Serial Number (SN)** of your xArm7? It is labeled on the robot base. We have done additional kinematics calibrations since last August and there is a way to extract and compensate the parameters to the URDF. I will check your SN to see if the compensation applies to your robot.

However, please note even after the compensation, the error can not be eliminated completely, we only guarantee the repeatability rather than absolute accuracy. Does your application require high physical accuracy?

vivekcdavid, 2024-03-30:
@penglongxiang 
I would prefer to share the serial number of the device in private? How can I reach out to directly?

Unfortunately we bought the machine before August. Is there a way to calibrate the machine and the URDF now?

penglongxiang, 2024-04-01:
Please contact us by email (support@ufactory.cc) and refer to this issue page in your message, and besides the SN, please also tell us your accuracy requirement and we will do some assessment first. Thank you.