xArm-Developer/xarm_ros #222: Force-Torque sensor has no data response

taisi5446, 2024-04-01:
I using the xARM6 with FT sensor and gripper

But when i using roslaunch of realHW with report_type rich or dev, there are any data on xarm/uf_ftsensor_ext_states or xarm/uf_ftsensor_raw_states. (only has 0 0 0 0 0 0, no data change when i push or tilt FT sensor.)

How can i solve this problem?

[image]

And, FT-sensor control box has wrong letter on Robot side. Actual Robot socket is Force Torque sensor, and actual Force Torque socket is Robot. (in Robot side(upside) of picture, right line is actual robot socket, and middle line is actual FT-sensor )

vimior, 2024-04-01:
@taisi5446 
1. Is the FT-sensor not enabled? Please refer to the #148 
2. Sorry, this sticker is an old model. I can confirm this is our latest design, the second head is for FT sensor, we will revise the sticker and user manual.

taisi5446, 2024-04-04:
thank you for your advice.
I solved the problem and question.

closed 2024-04-09 by vimior