xArm-Developer/xarm_ros #223: rviz disappeared using moveit

Lucaswaung, 2024-04-08:
### WSL Version
2.1.5.0
Ubuntu 20.04

### Repro Steps

when I run
`roslaunch xarm7_moveit_config xarm7_moveit_gazebo.launch`
rviz is running but there is nothing show in it.
[image]

another issue occurred when I run
`roslaunch xarm7_moveit_config realMove_exec.launch robot_ip:=<my_ip_address> [velocity_control:=false] [report_type:=normal]`
there is what happened, the program looped :
[INFO] [1712581059.659724, 0.000000]: Waiting for /clock to be available...
[INFO] [1712581059.660795, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller

I've tried `export LIBGL_ALWAYS_SOFTWARE=1` to prohibit GPU
Additionally, `roslaunch xarm_description xarm7_rviz_display.launch` will show the rviz UI.

penglongxiang, 2024-04-09:
Hi, please make sure you have followed the ReadMe and installed all necessary dependencies. Could you please also share the error or full information printed on terminal? We can not diagnose from the info provided so far.

Lucaswaung, 2024-04-09:
是的， 我按照步骤安装了所有依赖
并且我在使用
`roslaunch xarm_description xarm7_rviz_display.launch`
时，Rviz可以正常显示
但当我使用
`roslaunch xarm7_moveit_config demo.launch`
时，RVIZ无法正确打开，命令行中没有输出任何Error级信息，最近几行命令行输出如下：
`
[ INFO] [1712651641.464937827]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1712651641.464989326]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1712651644.628227331]: Loading robot model 'UF_ROBOT'...
[ INFO] [1712651644.628295330]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1712651644.669762834]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_<my_PC>/xarm7/kinematics_solver_attempts' from your configuration.
[ INFO] [1712651644.776839537]: Starting planning scene monitor
[ INFO] [1712651644.778221713]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1712651644.792538973]: Constructing new MoveGroup connection for group 'xarm7' in namespace ''
[ INFO] [1712651645.780859758]: Ready to take commands for planning group xarm7.
`

且当我使用
`roslaunch xarm7_moveit_config realMove_exec.launch robot_ip:=<my_ip_address> [velocity_control:=false] [report_type:=normal]`
命令尝试控制真实机械臂时：
命令行不断输出如下命令，无法控制：
[INFO] [1712581059.659724, 0.000000]: Waiting for /clock to be available...
[INFO] [1712581059.660795, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller

penglongxiang, 2024-04-11:
抱歉我们这边无法复现您描述的情况，请问你用的是xarm_ros最新的代码吗，下载之后有做任何修改吗？特别是.rviz后缀的文件。