xArm-Developer/xarm_ros #229: How can i solve the rosmsg index error on MATLAB2023b

OsitaRikuto, 2024-05-25:
Hello, here is RIKUTO.
I'm facing a problem right now.
I'm trying to manipulate xarm6 on matlab.
but the following error occured when generate ROS message using " rosgenmsg " function.
"The index exceeds the number of array elements. Index must not exceed 1."
[image]
The package used to generate the message is an official package cloned from this site.
I tried the command " clear rosgenmsg " and rosdep, but the error not solved.
Do you have any solution ? I'm in so much trouble.
On another robot arm's ros message, generation ros message is succeeded.
Does anyone have the same case?
Kindly await your help !!
thank you.

penglongxiang, 2024-05-27:
Hi @OsitaRikuto, **have you made any modifications** from our original code?  We have no experience with MATLAB ROS supporting functions, but I may infer from the error message and it seems there is a line in certain srv file which some (value) part is missing?  Also, have you tried the package can be successfully built in ROS **catkin** method?

OsitaRikuto, 2024-05-27:
Hi @penglongxiang 
Thank you reply,
I have no modifications from this official pakcage.
As a next measure, I would like to copy only the msg and srv directories to other new folders and verify whether this is the case with those folders.
If you don't mind, I would appreciate it if you could let me know if there have been any such cases in the past.
Thanks for your support, I appreciate it. Thanks.