xArm-Developer/xarm_ros #230: Unable to use moveit package when xarm is mounted on husky

rajsurya1012, 2024-06-05:
I have been trying to mount xarm6 on husky robot. I am able to successfully load the URDF on Gazebo and Rviz.
When I plan, a trajectory is generated but while execution, moveit fails.
However, the Gazebo arm moves.

Error:
[ERROR] [1717613146.781084954]: Semantic description is not specified for the same robot as the URDF.
[ INFO] [1717613152.237136092, 3.500000000]: Starting planning scene monitor
[ INFO] [1717613152.238330975, 3.501000000]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1717613152.254067081, 3.517000000]: Constructing new MoveGroup connection for group 'xarm6' in namespace ''
[ INFO] [1717613153.420308426, 4.682000000]: Ready to take commands for planning group xarm6.
[ INFO] [1717613303.644205440, 154.700000000]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1717613303.644286122, 154.700000000]: Planning attempt 1 of at most 1
[ INFO] [1717613303.646142762, 154.702000000]: Planner configuration 'xarm6[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1717613303.646903850, 154.703000000]: xarm6/xarm6[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1717613303.646931216, 154.703000000]: xarm6/xarm6[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1717613303.646957230, 154.703000000]: xarm6/xarm6[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1717613303.646981037, 154.703000000]: xarm6/xarm6[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1717613303.657519382, 154.713000000]: xarm6/xarm6[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1717613303.657565496, 154.713000000]: xarm6/xarm6[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1717613303.657705565, 154.713000000]: xarm6/xarm6[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1717613303.657863699, 154.714000000]: xarm6/xarm6[RRTConnect]: Created 5 states (3 start + 2 goal)
[ INFO] [1717613303.657950051, 154.714000000]: ParallelPlan::solve(): Solution found by one or more threads in 0.011380 seconds
[ INFO] [1717613303.660609744, 154.716000000]: SimpleSetup: Path simplification took 0.002573 seconds and changed from 4 to 2 states
[ WARN] [1717613305.187111063, 156.241000000]: Controller '' failed with error PATH_TOLERANCE_VIOLATED: joint5 path error 1.001089
[ WARN] [1717613305.187223361, 156.241000000]: Controller handle  reports status ABORTED
[ INFO] [1717613305.187284224, 156.241000000]: Completed trajectory execution with status ABORTED ...
[ INFO] [1717613305.197909249, 156.252000000]: ABORTED: CONTROL_FAILED

I have tried multiple ways to attach the arm including using "attach_to" argument, but everything leads to the same error.

penglongxiang, 2024-06-06:
Hi, the information provided so far is too limited for us to diagnose or reproduce your issue. Could you be more specific about which launch file you used and what you have changed from original code? Have you made same changes to both URDF and SRDF?

rajsurya1012, 2024-06-06:
Steps to reproduce error:
1. Install husky description.
2. In the xarm urdf, include husky description. Then change "attach_to" argument to "top_plate_link". This attaches the xarm on husky. There is not attach_to argument in the srdf file.
3. Launch this new xarm in empty world. In another terminal, launch xarm6_moveit_gazebo.launch.

When a target is set, a path is planned, but execution fails. However it is still executed on gazebo.

You will be able to reproduce the error using this.
Thank you!

penglongxiang, 2024-06-07:
Is there any **Error message** in the terminal after you launch the application?
Based on your description, I guess probably you did not load the ros-controller(**JointTrajectoryController**) for execution. Please refer to [xarm6_beside_table.launch](https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_gazebo/launch/xarm6_beside_table.launch) and add the controller part in your launch file. It is supposed to work together with xarm6_moveit_gazebo.launch in our demo as described [here](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#to-run-moveit-motion-planner-along-with-gazebo-simulator).