xArm-Developer/xarm_ros #234: handeye calibration problem

DonghanChen, 2024-07-12:
The poses that easy handeye generated for calibration can't get a good result. Therefore it is recommended to move the robot by our own.However, when I used ufactory studio to control xarm, ufactory will conflict moveit, but easy handeye need moveit to get the robot poses to compute the calibration result. It should be ok to use moveit goal request to control the xarm, but it is too hard to control it because handeye calibration also has some limitations in poses.
Could you find a way to fix it or make it more convenient?

penglongxiang, 2024-07-12:
@DonghanChen There is a [tip](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#722-precautions-for-hand-eye-calibration) here in the ReadMe, you need to specify `freehand_robot_movement:=true` as launch parameter to enable controlling with UF studio. Please follow these instructions and try again.

DonghanChen, 2024-07-15:
@penglongxiang 
Thank you for your reply, but it seems there may have been a misunderstanding. I would like to report a bug related to the hand-eye calibration process. EasyHandEye retrieves the robot's pose through MoveIt, but during free hand movements, UFactory's control directly causes MoveIt to crash. This results EasyHandEye unable to read each pose and got a crazy calibration result.
I have discussed this issue with your technical staff in wechat and got a positive  response. They advised me to raise this issue on GitHub.

penglongxiang, 2024-07-16:
Hi @DonghanChen Did you test the calibration process with **the latest code**? We have tried it today and the calibration can be done normally under control from UF studio. See the attached photo:
[image]

Moveit and other GUI did not crash, and the pose can be normally recorded by clicking "take sample" button. Please ignore the warning messages in yellow in the terminal under this situation.

DonghanChen, 2024-08-20:
@penglongxiang 
The bug disappeared after pulled the latest code, thanks a lot.

closed 2024-08-20 by DonghanChen