xArm-Developer/xarm_ros #235: xarm6 - isaac sim - usd file

Murali25111993, 2024-07-22:
Hello,

I am working on nvidia isaac sim, for that we have to provide usd file. 

Many leading robotic arm manufacturer are providing with usd file, I had trouble in converting the urdf to usd file. 

Please provide the usd file for all your product.

The main objective for this is to working with nvidia - isaac_ros_manipulator [link](https://curobo.org/tutorials/1_robot_configuration.html).

Thanks,

Murali25111993, 2024-07-24:
I have solved the issue. 

isaac_sim v 4.0.0

In case if anyone find difficulty in creating xrdf file. I am attching usd file , yaml file, xrdf file . 
[image]
[image]

[xarm6_1.zip](https://github.com/user-attachments/files/16358312/xarm6_1.zip)

Thanks

Murali

closed 2024-07-24 by Murali25111993

Oliverbansk, 2024-08-19:
Hi! Thank you for your great effort! Can I ask if you have the USD file of xArm5 or xArm7 robot?

m986883511, 2024-09-05:
[image]

can you control gripper use Gripper_Controller, I dont know how to set the params
[image]

Murali25111993, 2024-09-12:
No, I haven't tried using gripper. 

Thanks

23wjj, 2025-10-17:
> [image]
> 
> can you control gripper use Gripper_Controller, I dont know how to set the params [image]

Hello, may I ask if you have the USD file of xarm7 with the gripper?

inesdanjou, 2026-01-16:
Hello everyone, does anyone did a xArm7 usd file  with the bio_gripper ?
This file will really help me in my school project :)